revk/xESP32-Faikout #402: [FEATURE] Comfort features on CTXS15K3V1B

f4bl3d, 2024-07-24:
**Daikin hardware**
CTXS15K3V1B, Faikin connected on S403 (S21 wasn't available)

**Is your feature request related to a problem? Please describe.**
The following functions are not available on the Faikin web controls: Econo, Silent, Comfort-Sensor, Intelligent eye.
According to the device handbook / remote control, the device itself would be capable.

**Describe the solution you'd like**
At least Econo and Comfort-mode would be nice.

**Describe alternatives you've considered**
Alternatives: I'm really no expert, but with help could certainly provide an s21 dump through mqtt.

Thank you!

revk, 2024-07-25:
We have eco, it depends if you device responses to the relevant S21 messages.

Silent and comfort-sensor and intelligent eye are not, as far as anyone knows, available on S21.

If you believe there are S21 messages and responses for your device, let me know. Then I can add them.

f4bl3d, 2024-07-25:
Thank you for your fast response! The option for eco doesn't show up on the webcontrol screen. Does that mean that the device isn't capable for eco through s21?

revk, 2024-07-26:
Normally what shows is based on getting responses to S21 messages, so yes, usually, if not shown then it is not available.

It is all reverse engineered so possible we have missed some cases and alternate command.

f4bl3d, 2024-07-26:
Are these comms error messages add valuable information, or are they considered "normal" based on the revers engineered protocol?
I've cycled through eco and quite modes as these errors popped, but I can't say for certain, if they are related or just a coincidence. 

{"protocol":"S21","badsum":"DC","data":"02534930B4312B5C03"}
{"protocol":"S21","cmd":"RI","noack":true,"value":"16"}
{"protocol":"S21","timeout":true}
{"protocol":"S21","timeout":true}
{"protocol":"S21","badsum":"DA","data":"02473730FEFE30D603"}
{"protocol":"S21","cmd":"RN","noack":true,"value":"86"}
{"protocol":"S21","timeout":true}
{"protocol":"S21","badsum":"DD","data":"02534830B5322B5D03"}
{"protocol":"S21","timeout":true}
{"protocol":"S21","badsum":"F9","data":"025361303832AB7903"}

revk, 2024-07-26:
timeout and bad sum are not good, as communications errors, rather than something just not supported.

If they are rare, it may not be a problem, things get resent.

MassiPi, 2024-07-29:
> According to the device handbook / remote control, the device itself would be capable.

this is exactly the point.
there is no link between what the remote control can do and what the S21 can do. A lot of units have remote control functionalities (a common example is eco/powerful mode) that can't be set through S21 commands.

MuzzaMcD, 2024-07-29:
>  there is no link between what the remote control can do and what the S21 can do.

Correct. The IR Remote is a conundrum in itself!  The fact that there is 'logic' on board the remote makes the job of working out code harder than just looking at S21 comms. For example, if I set the comfort mode ON in the remote this will completely disable the fan button as logically Daikin is saying 'I'll give you comfort, stop playing with the fan!'
On my remote, ECO can't be set if the remote is in OFF mode. Also, the manual says ECO is not valid IF the power consumption is already low. How low? ... it don't say. 
.... just adds to the challenge eh? !!  :)

revk, 2024-08-18:
There is more work on S21 now, which may help with this in due course.