revk/xESP32-Faikout #406: [BUG] Unit switching from cooling to heat after approx 40 mins on its own

urbozz, 2024-07-29:
Using the Faikin from Amazon, on a FVXG25K2V1B using S21, the unit is set to Cool using HomeAssistant (so via MQTT) after a period of time, the unit sets itself to heat. I have been able to replicate the issue consistently. Appears to have start happening after turning off the Faikin auto mode. Log from HomeAssistant showing the unit setting Heat. Let me know what diagnosis may be useful.

[image]

revk, 2024-07-30:
Can you screenshot the control page on the Faikin?

urbozz, 2024-07-30:
While in Cool - just turned on, will take another once switched to heat.
[image]

revk, 2024-07-30:
Somewhat at a loss, there is some MQTT logging from the Faikin, which may help, if you can capture it.

With the Faikin auto off, and the timed/auto off, it should indeed not change by itself.

urbozz, 2024-08-04:
I power cycled the whole system, and since then cannot replicate the issue, suggesting it may have been the Daikin itself setting itself to heat.

closed 2024-08-04 by urbozz