revk/xESP32-Faikout #409: [BUG] Mode controls by HA don't work with CN_WIRED

Sonic-Amiga, 2024-07-30:
Continuation of https://github.com/revk/ESP32-Faikin/issues/240#issuecomment-2257166442

type-rke, 2024-07-30:
I don't know if this is what you need:
[mqtt-8721a2dc4c281e43ba86ed5ee120532f-aircoliving-e147c2f99dabdcdb505688ffef4ce606 (1).json](https://github.com/user-attachments/files/16432819/mqtt-8721a2dc4c281e43ba86ed5ee120532f-aircoliving-e147c2f99dabdcdb505688ffef4ce606.1.json)

if not, how can i extract the log ?

Sonic-Amiga, 2024-07-30:
This is something from HA and this doesn't give me any info. I meant MQTT log.
Procedure is described here: https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/12#issuecomment-2192696938
I am interested in MQTT messages, which HA sends to Faikin when you try changing mode.

type-rke, 2024-08-01:
i can't see any message send to Faikin, only the ones from the cooling / fan modes are beeing send
[airco_log.txt](https://github.com/user-attachments/files/16460129/airco_log.txt)

Sonic-Amiga, 2024-08-04:
You mean - turn off fixstatus ?

Sonic-Amiga, 2024-08-04:
By the way, please list your A/C type in https://github.com/revk/ESP32-Faikin/wiki/List-of-confirmed-working-air-con-units .

Sonic-Amiga, 2024-08-05:
@type-rke You didn't answer. The issue is fixed when fixstatus is on or off ?

Sonic-Amiga, 2024-08-05:
@revk Regardless, this setting only affects Faikin's own status reporting, it should't affect HA communication. Side effect ?

Sonic-Amiga, 2024-08-05:
BTW, @revk where can i find documentation on MQTT messages HA understands ? How do you code support ? I am browsing their web site; they're talking about some YAML files, but there's nothing regarding mqtt.

Sonic-Amiga, 2024-08-05:
I found docs. Yet, HA communication should not be affected.

HA status us reported under setting/[hostname]; Faikin's own reporting (which is affected by fixstatus) comes under Faikin/[hostname] ... I am confused.

@type-rke Can you show to us topicXXX settings in your config (Advanced page) ? Maybe something is messed up there...

revk, 2024-08-06:
I moved the HA stuff in to the general `state` message to reduce the number of topics it had to subscribe to and allow it to see the `up` for availability.

With default topics there should be

`state/nameoffaikin` which is what HA listens to, and includes `up` and a load of current values.

`state/nameoffaikin/status` which is also a regular status report for MQTT, not for HA use. Yes, this is basically duplicating the HA stuff - may tidy that up some time.

`Faikin/nameoffaikin` which is a periodic report with ranges of values. I think this is what `fixstatus` impacts.

The exact format will depend on the topic related settings though.

type-rke, 2024-08-06:
> @type-rke You didn't answer. The issue is fixed when fixstatus is on or off ?

I'm sorry, I'm on vacation for 2 weeks now, I check when I'm back.  for me everything is working now, but I can't access the seating page at the moment.

I can control the airco and thé mqtt HA settings are also ok

Sonic-Amiga, 2024-08-06:
@type-rke Okay, happy vacation
@revk I am thinking that perhaps there's some bug in HA, which is triggered by similarity of topics: state/nameoffaikin vs Faikin/nameoffaikin . Because only that somehow explains what's happening

Looks like some of us has to go through HA learning curve, install it and play with it.

revk, 2024-08-07:
I have HA here now for testing, but need to reinstall it. I'll get around to it.

From what I have seen, no way it should be confused by topics.

However the MQTT settings in HA, selecting the device, and a "..." menu thing, gets you a really good diagnostic showing the config and the messages it has seen and sent.

That should help diagnose the issue.

Sonic-Amiga, 2024-08-07:
Okay, thank you. Frankly speaking i am busy with other things, including taiming my BRP. I want to finally reverse engineer how it handles night fan speed and finds out AC features. Albeit the latter can be just hardcoded into firmware, hence they have tons of BRP model numbers.

revk, 2024-08-18:
I'm assuming we have this sorted now, let me know if not.

closed 2024-08-18 by revk