revk/xESP32-Faikout #410: [FEATURE] Disable code signing

dwmw2, 2024-07-31:
**Daikin hardware**
No.

**Is your feature request related to a problem? Please describe.**
I wanted to install esphome on my Faikin module to use it with a Mitsubishi Ecodan ASHP.

**Describe the solution you'd like**
I'd like a way to buy the preinstalled unit from e.g. Amazon but then load esphome (or even my own builds of this project) onto it. I didn't try, but I understand that the code signing prevents that.

**Describe alternatives you've considered**
I hooked up the serial port by sticking pins in the holes and not wobbling them too much, manually shorted IO0 to ground and plugged/unplugged the 5v wire. It worked, but I didn't like documenting it like that for others to follow.

**Additional context**
https://github.com/gekkekoe/esphome-ecodan-hp/discussions/31

dwmw2, 2024-07-31:
Editing to make it clear that I don't want to disable it by *default*, necessarily. Just have a sane way for the user to disable it.

revk, 2024-07-31:
I have deliberately not made the code signing hard set in the chip, so you can simply load a new bootloader (with with no signing or you own signing) using a serial lead.

You managed it? https://github.com/gekkekoe/esphome-ecodan-hp/discussions/31

revk, 2024-08-18:
Basically, the only way is replace the bootloader, and that is done over the serial/USB. Allowing it by any other means creates a security issue. Physical access creates enough of a security barrier (in my view).

closed 2024-08-18 by revk