revk/xESP32-Faikout #411: [BUG] cloudflared proxied faikin not working

jsapede, 2024-07-31:
**Faikin hardware**
Please advise if this is a Faikin you purchased (e.g. from Amazon), had built, or some other device.
Ideally, please include picture of both sides of the board itself as this will help confirm the version.

**Daikin hardware**
Please advise the model of your Daikin, and how you are connected (e.g. S21 lead, X50 lead, etc).

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

revk, 2024-07-31:
Sounds like something is adding (long) header fields?

jsapede, 2024-07-31:
I dont really know, never had problem on any apps with those tunnels witch acts like a proxy over vpn

revk, 2024-08-01:
Well it is a tiny microcontroller and allocates some static buffers in the http server handling. So the config needs to know how big.

It would help to know what headers and how big!

jsapede, 2024-08-01:
> Well it is a tiny microcontroller and allocates some static buffers in the http server handling. So the config needs to know how big.
> 
> It would help to know what headers and how big!

Ixhave non Idea how to proceed but if you need some logs i Can have a try

revk, 2024-08-01:
I don't know either, the http server is part of the ESP IDF, so not sure if I can get any logs, and if I could it would be over a serial debug port.

Do you have anything else you can access in the same way that allows you to dump/report what headers are sent?

Or can you packet dump on your LAN (this is http not https so should be clear).

revk, 2024-08-18:
If you are able to find headers being sent, but setting something else to access by proxy, then that may allow me to address the issue. Re-open if/when you do.

closed 2024-08-18 by revk

jsapede, 2024-12-07:
Ok now in winter in thebhouse with the hvac i made some extra tests 

Accession the main page through cloudflared tunnel look ok

[image]

When i go to the settings page got error 

[image]

But, if i directly go to the settings page with its URL it works 

[image]

In this page if click on the various pages links, it shows the same error 

Reverting back to the main page exhibit same behaviour. Clicking on the link AT page bottom rises error while using directly homepage URL works. Looks like some referers or ws could be source of the problem.

revk, 2024-12-07:
Yeah, it is adding extra headings that are too long.

jsapede, 2024-12-07:
Investigating further with développement console shows error is coming from ws requested over HTTPS
 
Reading ai explanations in the chrome console tells to modify settings to allow wss connection.

Moreover it only returns errors in chrome. Using Firefox poses no problems

jsapede, 2024-12-07:
[image]

jsapede, 2024-12-07:
[image]

revk, 2024-12-08:
Given the the control/web page is not https it makes sense that we would use ws, not wss, for the web socket.

This really is a tad silly, you are wrapping it in https and expecting it to work.

jsapede, 2024-12-08:
Well, the initial problem is that the climate control heating of a holyday house and used to heat it 1 or 2 days before we arrive

As house use over the year, connection IS made with low cost 4G mobile router. As most of 4G connections it uses CGNAT with no ability to redirect ports

The only way found to Access it was to use cloudflared tunnel, witch IS a sort of hosted proxy combined with vpn. Cant even try with a simple proxy as the connection doesnt have a "Real" IP.

With many others apps involving ws (homeassistant, immich, vscode ...) it usually works flawlessly having HTTPS connections on the cloudflared proxy.

revk, 2024-12-08:
OK, the code does handle HTTPS already.

```
                  "ws=new WebSocket((location.protocol=='https:'?'wss:':'ws:')+'//'+window.location.host+'/status');"   //
```

So not sure why your browser is not doing that. `location.protocol` should be `"https:"` so it should use `wss:`

jsapede, 2024-12-08:
Ok thx, will try to have a look with the security settings maybe 
Thx for the help