revk/xESP32-Faikout #418: [FEATURE] Pullup resistor on RX line

dwmw2, 2024-08-09:
**Daikin hardware**
Mitsubishi Ecodan

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

The Mitsubishi units don't seem to have their own pullup on their TX line.

**Describe the solution you'd like**
Please add another 10kΩ resistor to pull up pin 2 of J1 to pin 1.

**Describe alternatives you've considered**
I used an external one. It is messy.

revk, 2024-08-09:
A second pull up would pull Tx to Rx if no 5V line!

dwmw2, 2024-08-09:
And a diode or two then?

dwmw2, 2024-08-09:
Anything but this:
[image]

revk, 2024-08-09:
I mean maybe diodes. Pain in the arse when so many just work.

dwmw2, 2024-08-09:
True, but it would be amazing if it worked out of the box for the Mitsubishi units too.

revk, 2024-08-10:
A diode will introduce a drop which is a problem. I suspect it has to be a FET or something. I'll have to see if possible. But not ideal.

revk, 2024-08-10:
One thing we could do, and you could test.

We pull up Tx, the issue is Rx, which presumably they are only open-drain driving, hence needing a pull up.

We could pull that up to 3.3V, can you try that?

The only issue is if they drive at 5V on some models. At 10k would mean 200uA driving to the 3.3V supply, which I doubt would be a problem. So we can probably safely add a 10k on the Rx side to our 3.3V rail, and that would keep it independent of the Tx and (optional) 5V rail.

revk, 2024-08-10:
OK I can make a small batch with a pull up - but we need to test on this model and others. It will take a week to arrive.

revk, 2024-08-10:
I'll need some addresses for sending some test boards. Email in to A&A FOA me (Adrian).

closed 2024-08-10 by revk

dwmw2, 2024-08-10:
Sounds good, thanks!

dwmw2, 2024-08-10:
Perhaps it ought to be robust enough even for 12V on the RX line, even if that happens by accident?

On the quad FTDI board which was also working with this heat pump, its unconnected RX line floats at 2.4V — about a 0.7V drop from 3.3V?

2.5V ought to be perfectly sufficient; it's enough for the Faikin board itself to be able to tell the difference between floating high, and something external pulling it down to ground.

How about a diode with the pullup so that nothing at all can feed in to the chip power rail?

revk, 2024-08-10:
I'd be surprised at anything not 5 or 3.3.

dwmw2, 2024-08-10:
I dunno. I just found a better 10kΩ resistor to put directly in the Dupont plug and tried plugging it in, and it didn't work (didn't power up at all) Probably got the five pins in the wrong order, and 12V ended up *somewhere* it shouldn't... (still works with old cable pictured above though, so no harm done).

dwmw2, 2024-08-22:
The units with the built-in 10kΩ pullup work well on the Mitsubishi Ecodan heat pumps, thanks!

revk, 2024-08-23:
Perfect. I should have some with a diode soon just to be neater.

dwmw2, 2024-08-23:
Makes sense.

_"Never assume the user won't apply 12 volts to a pin which didn't expect it"_ is certainly a good maxim to live by.