revk/xESP32-Faikout #422: [BUG] "make update" over-updates SQLib

Sonic-Amiga, 2024-08-10:
"make update" executes:

git submodule update --init --recursive --remote

This pulls latest versions of all submodules, including sub-sub-modules. This ends up in pull-ing SQLib/stringdecimal. But SQLib itself has a super-project-pointer, which still sits at 8f121f25da097e0673b292f038f7600df1f2eb45, and that has never been updated, so i end up in "dirty" status for the repo untill i manually checkout stringdecimal to what it's expected to be

revk, 2024-08-10:
Ok I have literally no clue, sorry.

I struggle with git at the best of times.

Tell me the "fix"

Sonic-Amiga, 2024-08-10:
Oookay, i'll try to come up with one. Standby :)

Sonic-Amiga, 2024-08-10:
Let's try https://github.com/revk/ESP32-Faikin/pull/423

For me it at least didn't clobber sub-sub-modules, top-level subs are already up to date for me, so full testing waits till next time.

closed 2024-08-13 by revk