revk/xESP32-Faikout #424: MQTT Current Temperature

Cees-tech, 2024-08-10:
Hello, I have a question about the current temperature being send over MQTT in 'state/Faikin ESP32/status/home' 

This data much be handled by [MQTTThing](https://github.com/arachnetech/homebridge-mqttthing)
I'm using these topics in MQTTThing which are all working except 'Get Current Temperature':
[image]

I get this error:
`[homebridge-mqttthing] This plugin generated a warning from the characteristic 'Current Temperature': characteristic was supplied illegal value: number 0 exceeded minimum of 18. See https://homebridge.io/w/JtMGR for more info.`

I think that MQTTThing has a problem with the number format of the temperature  but I can't figure out what's wrong here. Off course the problem could be at MQTTThing but I thought its better to ask here first.
Hopefully anyone can help me with this? 
Thanks in advance, Cees

Sonic-Amiga, 2024-08-10:
Hmm... Current temperature of 0 degrees feels strange...

revk, 2024-08-11:
A topic like state/name/status/temp is not sensible. The topic would be state/name or state/name/status and then a json payload with a field called temp.

Also the "status" usually outputs an array of min/ave/max which you probably want a fixed single value.

As said, the fixstatus setting may help.

Cees-tech, 2024-08-12:
I turned "fixstatus" on. 
I also noticed that the topic **state/Faikin ESP32/status** doesn't update at all. Therefore I also turned "homeassistent" on and grabbed the required data from **state/Faikin ESP32/ha**

> A topic like state/name/status/temp is not sensible. The topic would be state/name or state/name/status and then a json payload with a field called temp.

Your'e right, I'm still learning MQTT and JSON. If I read the documentation of MQTTThing correctly is supposed to be like this (with $. sign):

```
"type": "heaterCooler",
            "name": "Airco Woonkamer",
            "topics": {
                "getOnline": "state/Faikin ESP32/ha$.online",
                "getRotationMode": "state/Faikin ESP32/ha$.fan",
                "setRotationMode": "command/Faikin ESP32/fan",
                "getActive": "state/Faikin ESP32/status$.power",
                "setActive": "command/Faikin ESP32/power",
                "getCoolingThresholdTemperature": "state/Faikin ESP32/ha$.target",
                "setCoolingThresholdTemperature": "command/Faikin ESP32/temp",
                "getCurrentHeaterCoolerState": "state/Faikin ESP32/status$.mode",
                "getCurrentTemperature": "state/Faikin ESP32/ha$.temp",
                "getHeatingThresholdTemperature": "state/Faikin ESP32/ha$.target",
                "setHeatingThresholdTemperature": "command/Faikin ESP32/temp",
                "getTargetHeaterCoolerState": "state/Faikin ESP32/status$.mode",
                "setTargetHeaterCoolerState": "command/Faikin ESP32/mode"
            },
            "onValue": "1",
            "offValue": "0",
            "minTemperature": 18,
            "maxTemperature": 33,
            "manufacturer": "Daikin",
            "accessory": "mqttthing"
```

Is it correct that commands can be send in seperate topics like "command/Faikin ESP32/temp" and "command/Faikin ESP32/mode"?

With the configuration above Homekit gets the measured temperature, I can turn the AC on/off, set modes and temperature. So everything I need is working now!

revk, 2024-08-12:
Try latest beta code please as `/ha` is changing.

I'll make a release in fact...

njh, 2024-08-13:
@Cees-tech it would be really good to get a Wiki page setup with Homebridge instructions on it.
I had a go a while ago but didn't manage to get the fan control settings sorted out.

I am currently using [homebridge-daikin-local](https://github.com/cbrandlehner/homebridge-daikin-local), which works over the emulated HTTP API. But I like the idea of it being event driven over MQTT.

revk, 2024-08-13:
You should be able to see the MQTT messages. HA on will send the main status message, but I suggest ensuring to to date.

MQTT is likely a lot better than the legacy APIs, yes.

revk, 2024-08-18:
As I say try latest code. It is released now using `state/nameoffaikin` and no `/ha` on the end.

closed 2024-08-18 by revk