revk/xESP32-Faikout #428: [BUG] Forgot mqtt and login settings after auto ota update

bigmoby, 2024-08-12:
**Faikin hardware**
This is a Faikin purchased from Amazon. (Faikin-S3-MINI-N4-R2: b16bfc4 2024-08-12T14:02:04 S21)

**Describe the bug**
After a clean Faikin installation (through dupont wires...) and a plain first-access in order to define ssid, wifi password, all mqtt settings and password access, I could access to the settings pages and it was perfectly fine. After that the device automagically updates itself and once it rebooted I wasn't able to access Settings page: "wrong login password" and also the mqtt settings are wrong. Now, I think, I am forced to reset the device.

**To Reproduce**
Steps to reproduce the behavior:
1. First installation and successful Settings page access
2. Auto OTA update
3. Unable to access to Setting page because wrong login password and also mqtt settings are lost.

**Expected behavior**
After the automagic firmware update all setting should be maintained

**Additional context**
How I could reset the Faikin device?

revk, 2024-08-13:
Do you have any screenshots? This does not seems to make a lot of sense.

At present the only way to factory reset is using a usb lead to pads on the back.

But I need to understand how it has got in to this state. Unless you explicitly filled in the password field in settings, it should not need a password!

bigmoby, 2024-08-13:
Yes, sure, these are the screenshots...I don't know if useful, I don't think so.
As described in the issue, after the first boot I applied the wifi, mqtt and login settings. After that the device has been updated (autoupdate) and rebooted and missed all the settings but the wifi settings remains working.
Thank you for your help!

[image]

[image]

revk, 2024-08-13:
Ok that only makes sense if you set a password.

I'll have to work out a factory reset, assuming you don't have a lead and tools to do yourself.

bigmoby, 2024-08-13:
> Ok that only makes sense if you set a password.

Yes, I've applied the password (...and there's no mistake...I remember it 🙂) but now it is identified as wrong.

> I'll have to work out a factory reset, assuming you don't have a lead and tools to do yourself.

I have a USB to serial converter (like this 
https://amzn.eu/d/5VDxz50 ) that I use to "tasmotize" sonoff relay. Is it useful in order to reset or flash?

revk, 2024-08-14:
You can either connect to 4 pins for USB or the 5 pins for serial and GPIO.

There are flash instructions somewhere in the documents, but you want to erase flash first to clear the stored settings including password.

bigmoby, 2024-08-14:
Thank you @revk for your support. Please, could you give me some directions or photos of the contacts I will have to create between the converter and the faikin?

revk, 2024-08-14:
Does this help?

[image]

revk, 2024-08-14:
I believe it usually gets connected to DTR on a serial lead for that purpose, but yes, if shorted to GND it will go in to boot mode on power up.

bigmoby, 2024-08-15:
Hi @revk , I tried to reset but when the Faikin is reconnected to the AC nothing happen...neither the menu! Is it bricked?!

[image]

revk, 2024-08-15:
You erase flash, so it is bricked, yes, obviously.

Now reflash the code.

bigmoby, 2024-08-15:
Ok perfect, should I clone the project and recompile by myself or could I download a builded firmware?
Thanks

revk, 2024-08-15:
There are some instructions on loading the binaries from the project release directory, that is probably best.

If you build yourself, you would have your own signing key and it would not auto update in future.

But up to you.

bigmoby, 2024-08-15:
Great! Thank you for you support @revk 🙌
I'll provide to create a PR in order to improve the README and add a section for reset and re-flash the device, if you want.

[image]

bigmoby, 2024-08-15:
Hi @revk , I create this PR https://github.com/revk/ESP32-Faikin/pull/438 
Thanks again for your support 👍

revk, 2024-08-15:
No problem, glad it is all sorted.

closed 2024-08-18 by revk