revk/xESP32-Faikout #429: [BUG] Faikin unable to boot

dandwhelan, 2024-08-12:
**Faikin hardware**

[image]
[image]

**Daikin hardware**
I've a Daikin Heatpump Altherma Monoblock EDLA06E2V3

**Describe the bug**
I've purchased two Faikin units from Amazon, One was a few months ago and the second was yesterday in hope there was an issue with the one I purchased a few months ago. I have tried both on my headpump using the below instructions and I cannot get the Faikin to startup, flash, light up or broadcast a WiFi AP.

I've followed two guides. One was a Youtuber with visual instructions and a second was a document I found on how to wire the device into my headpump.

https://raomin.github.io/ESPAltherma/
https://www.speaktothegeek.co.uk/2024/03/daikin-altherma-3-heat-pump-and-home-assistant/

**Expected behavior**
I expected to plug this in, lights to start flashing and a WiFi AP becomes available.

**Additional context**
I have this USB Device which I was hoping could emulate power going to the device so I can see it booting. I'm not sure if this can be helpful for troubleshooting. 
I am using Female to Female jumper cables which I use for other ESPhome projects. I dont have a special case as I didnt think I'd need one. 

[image]

Any help would be appreciated! Thanks

Sonic-Amiga, 2024-08-12:
From your first photo i see that power (12V) is not connected. That's the power pin, not "+5V"
What is connector on AC side ?

dandwhelan, 2024-08-13:
[image]

revk, 2024-08-13:
Yeh, you seem to have connect all except the power and then said it won't power up.

revk, 2024-08-13:
That also looks like X10A not S21, is it the same?

dandwhelan, 2024-08-13:
I have to use X10a due to the being no s21
I can try it with all 5 pins connected, but none of the guides seem to use pin 4 with my heatpump

dandwhelan, 2024-08-13:
OK ill give it a try with all the pins connected. I was going off the below picture, as I couldn't find much documentation around the x10a. 

[image]

revk, 2024-08-13:
Ok if X10A has nothing on pin 4, then run the pin 1 5V from it to pin 4 on the Daikin.

It is designed for S21. On a real S21 connector pin 1 is either NC, or 5V but too low current to run an ESP.

revk, 2024-08-13:
OK you can make a lot less noisy. Do you know if the X10A is compatible with S21?

You can change settings, set `nox50a` and `noswaptx` and `noswapx` that should try just S21 protocol.

The logs should be cleaner then.

If still no joy then, turn it all off and back on, it may be the aircon is confused by the attempts at alternatives.

dandwhelan, 2024-08-13:
Unfortunately more of the same 

13/08/2024 10:50:39(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:37(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:35(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:33(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:31(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FC"}
13/08/2024 10:50:29(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:27(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:25(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:23(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:21(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FC"}
13/08/2024 10:50:19(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:17(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:15(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:13(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:11(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:09(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FF"}
13/08/2024 10:50:07(-1.99 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FC"}
13/08/2024 10:50:05(-2.01 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:03(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:50:01(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:59(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:57(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FC"}
13/08/2024 10:49:55(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:53(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:51(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:49(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:47(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:45(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:43(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:41(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:39(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:37(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:35(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:49:33(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
[... 95 lines cut ...]
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:55(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:53(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:51(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FC"}
13/08/2024 10:47:49(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:47(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:45(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:43(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:41(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:39(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FF"}
13/08/2024 10:47:37(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FF"}
13/08/2024 10:47:35(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
13/08/2024 10:47:33(-2 seconds) 
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}

revk, 2024-08-13:
That does suggest that maybe it is not S21...

Do you have any reference for X10A being the same. Seems odd it would be different, but at this point I'd be looking at an oscilloscope. May be worth starting a discussion thread as a lot of users are on here and someone may know.

dandwhelan, 2024-08-13:
Ok thanks for your help :)

closed 2024-08-13 by dandwhelan

revk, 2024-08-13:
It may also be worth connecting pin 1 to pin 1 and 4, not just pin 1, if possible.

dandwhelan, 2024-08-13:
I tried that out but no joy

revk, 2024-08-13:
Is it connected to MQTT at all? If so we can do a factory reset.

If not, I'll have to work out a change to allow a factory reset for you.

Suffice to say, latest code is not causing this for other people. So all very odd.

revk, 2024-08-13:
Sorry replied to wrong ticket!

revk, 2024-08-13:
Worth trying to find anyone else that has used X10A basically. See if they had success.

Sonic-Amiga, 2024-08-15:
@revk The picture he posted says "Altherma", and some time ago you added some stubs for this protocol. Seems you've got the first user to test it.
The picture seems to be taken from https://raomin.github.io/ESPAltherma/ ; so you may check their implementation.
However, seems Altherma is crappy; it only allows for monitoring and very basic on-off control via external relay, simulating a thermostat. As far as i can conclude from their description.

revk, 2024-08-16:
Indeed, do let me know what works and does not work, and we can maybe get some dumps of what it sends.

dandwhelan, 2024-08-16:
OK, as the heatpump is outside and the panels are awkward too fit I only
want to be doing this once or twice a week.

Can you please how to get a dump just in case it didn't work.

revk, 2024-08-16:
There is a `dump` setting. It writes stuff to MQTT.

Ideally you try different settings and so on, and report what he MQTT dump shows and what the settings are so we can see which bits do what and hence ensure we can report it all.

dandwhelan, 2024-08-16:
Nice one. I'll take a look today.
I'll keep the faikin unit plugged in for more testing.