revk/xESP32-Faikout #431: [FEATURE] Expose more switches to HA

syphernl, 2024-08-13:
With the latest stable version of 2024-08-12 the new `haswitches` is introduced which exposes switches for power and the streamer to Home Assistant. Can this be expanded to also expose switches such as:

- Econo mode
- Powerful mode
- Comfort Mode
- Sensor Mode
- Quiet Outdoor

revk, 2024-08-13:
Always the way, feature creep, LOL.

- Do we have sensor mode, I did not think we did (i.e. not in S21)
- Do we have outdoor quiet mode?

But extra switches are something that could be added.

I may end up making a more general HA management library first as we have calls for HA integration on a number of other projects as well.

revk, 2024-08-13:
Oh, OK, sorry, I don't recall getting sensor more working - I have a feeling my unit does not do it, which is probably why.

But yes, these could be added as ha switches.

revk, 2024-08-13:
Sorry, silly question, what HA icon codes would be appropriate for these?

syphernl, 2024-08-13:
> Sorry, silly question, what HA icon codes would be appropriate for these?

I think these icons would probably be best suited:

- Econo mode:  [mdi:trending-down](https://pictogrammers.com/library/mdi/icon/trending-down/)
- Powerful mode: [mdi:arm-flex](https://pictogrammers.com/library/mdi/icon/arm-flex/)
- Comfort Mode: [mdi:teddy-bear](https://pictogrammers.com/library/mdi/icon/teddy-bear/)
- Sensor Mode:  [mdi:motion-sensor](https://pictogrammers.com/library/mdi/icon/motion-sensor/)
- Quiet Outdoor: [mdi:volume-minus](https://pictogrammers.com/library/mdi/icon/volume-minus/)

revk, 2024-08-13:
[image]

Try the latest beta in a few minutes...

closed 2024-08-13 by revk

syphernl, 2024-08-13:
Cool! Thanks for implementing this so quickly.

syphernl, 2024-08-13:
Works like a charm, thanks!

sm-Fifteen, 2024-08-21:
> I think these icons would probably be best suited:
>
>     * Econo mode:  [mdi:trending-down](https://pictogrammers.com/library/mdi/icon/trending-down/)
>
>     * Powerful mode: [...]

May I suggest [mdi:home-battery](https://pictogrammers.com/library/mdi/icon/home-battery/) for Econo mode, given it's meant to limit power draw so weak breakers don't trip when other power hungry devices are active, or for users who are *very* power concious, like when electricity cost is high? It's a small detail, but I figure it would be more evocative than some  "📉 stonks" chart.

revk, 2024-08-22:
Fine by me, not sure if any changes today as another big project on.