revk/xESP32-Faikout #434: No voltage on PIN4

Take-Off90, 2024-08-13:
Hardware: Daikin Emura 3 (FTXJ35A2V1BB)

Problem: No power on PIN4 of S21 port
--> Faikin has no power

I tried to connect PIN1 and PIN4 to power the Faikin with 5V, but I think there is not enough power to start the Faikin.

I tried to use an external power supply on PIN4 and PIN5. Faikin is now working, but no connection to the AC.
I think the issue here is that Faikin/Daikin is using a common ground on PIN5.
I'm a bit afraid of connecting PIN5 of Faikin with PIN5 of S21 and GND of the external power supply.

Is there any chance to power the Faikin directly from my AC?

revk, 2024-08-14:
We'll be careful.

But no, it won't work without GND.

If your power supply is isolated, and if a proper S21 (not X403), then probably OK

Take-Off90, 2024-08-14:
Do you know whats the reason behind the "no voltage" on PIN4. Any chance to activate this?
Are there any other possibilities to connect the Faikin?

The external power supply can't be a final solution. I hate that Daikin is making it so hard to create local access.

revk, 2024-08-14:
All the S21 I have seen have 12V(ish) on pin 4 with enough power for the Faikin. If you have, I think, an X10A, I gather that has nothing on pin 4 and a bit more power on pin 1 (5V). No idea why. But I doubt you can change it.

Only other thing is if there is a separate power feed from another connector maybe.

But always take a lot of care with anything mains powered, at your own risk!

yufaxt, 2024-08-14:
where did you plug it to? for me it works with the s21 socket on top (marked in red)
[image]

Take-Off90, 2024-08-14:
Thats the same port I tried. S21
Do you have power on PIN4? What is the product code of your AC?

revk, 2024-08-14:
My understanding is power on pin 4 is normal for any S21 connector.

Take-Off90, 2024-08-14:
> My understanding is power on pin 4 is normal for any S21 connector.

Mine too, but unfortunately no :( On google you can find a few threads where people have the same problem. Maybe Daikin changed something.

yufaxt, 2024-08-14:
> Thats the same port I tried. S21 Do you have power on PIN4? What is the product code of your AC?

FTXJ35A2V1BW

Take-Off90, 2024-08-14:
> FTXJ35A2V1BW
    FTXJ35A2V1BB is mine. I don't know what the difference between those two models is.

Maybe there is something not working correctly on my AC

yufaxt, 2024-08-14:
> > FTXJ35A2V1BW
> > FTXJ35A2V1BB is mine. I don't know what the difference between those two models is.
> 
> Maybe there is something not working correctly on my AC

Just the color afaik, White vs Black, I assume

Sonic-Amiga, 2024-08-15:
> Problem: No power on PIN4 of S21 port

Are you sure ??? That's weird. As far as i can see it's Daikin's normal way to power plug-in boards. Pinout can be different though. Have you checked all the pins with a multimeter relative to some known GND ?

Take-Off90, 2024-08-15:
> As far as i can see it's Daikin's normal way to power plug-in boards

That's what I expected. 
I checked all the PIN's with a multimeter bevor connection. Known GND was PIN5

We have two indoor units, and I will try it on the other one. If it works there, the other unit is defective. If it is the same issue the problem continues :-(

Sonic-Amiga, 2024-08-15:
If we turn on our imagination; we may think of Daikin removing some components from board for "cheap" A/C variants, which are not supposed to be connected to wi-fi.
What does the official site say ? Is there a wi-fi accessory for this particular model ? If yes, would be interesting what it looks like.

Take-Off90, 2024-08-16:
This model has the "new" Wifi Module from Daikin which only supports the cloud. 
Another user was successful connecting the Faikin and used it parallel to the Daikin modul. 

I will check on my other unit if it has the same issue. As you all said, there needs to be power on PIN4 as you can connect nothing on this port without power supply.

Sonic-Amiga, 2024-08-16:
> new" Wifi Module from Daikin

1. That's damn interesting. What does it look like ? Can you make high-quality photo, or maybe even scan, of both sides of PCB ?
2. Are you sure it's cloud-only ? I'm curious. It's easy to check whether it has local API:
- Locate its IP address; if you know MAC, you can do it by checking DHCP status of your router
- If you don't know MAC, you can figure it out by physically switching A/C power off and on. You'll see who appears and disappears in your router status page
- Go to http://ip_address_you_found/aircon/get_model_info . If it replies, the controller still has the local API. And if so, you;re welcome to S21 reverse engineering topic; i may eventually ask you to run some experiments with this controller and our A/C simulator. We're interested in protocol details.
-

yufaxt, 2024-08-16:
It’s BRP069C4x. I didn’t try much to locate it physically but it wasn’t close to the board. In any case I did attempt http://ip_address_you_found/aircon/get_model_info and nmap to see if it even listens on 80 and it doesn’t. Cloud only it seems

revk, 2024-08-17:
This all started because my a/c had a cloud only WiFi module.

Take-Off90, 2024-08-17:
> * Go to http://ip_address_you_found/aircon/get_model_info . If it replies, the controller still has the local API. And if so, you;re welcome to S21 reverse engineering topic; i may eventually ask [...]

Hey Sonic,

I already tired this, as I read the other post from you. But no success.  

I can try to make a photo of it next time I open my A/C. Maybe someone else has already done one. 

But it sure is cloud only. I contacted Daikin because of this issue and they said it is like it is. This desicion is made by Daikin Europe, and Daikin Germany can do nothing about it.

yufaxt, 2024-08-17:
> > > 
> > 
> > 
> > ```
> > * Go to http://ip_address_you_found/aircon/get_model_info . If it replies, the controller still has the local API. And if so, you;re welcome to S21 reverse engineering topic; i may eventually ask you to run some experiments with this controller and our A/C simulator. We're interested in protocol details.
> > ```
> 
> Hey Sonic,
> 
> I already tried this, as I read the other post from you. But no success.
> 
> I can try to make a photo of it next time I open my A/C. Maybe someone else has already done one.
> 
> But it sure is cloud only. I contacted Daikin because of this issue and they said it is like it is. This desicion is made by Daikin Europe, and Daikin Germany can do nothing about it.

Please do take a photo, I regret forgetting to do so :( I also wanted to take one of where I placed the faikin eventually (below the boards, but still outside of the metal protected area). Unfortunately the case I printed wouldn’t fit but the pcb itself did so I left it as is. just curious if others found a better placement

Sonic-Amiga, 2024-08-17:
So it's officially cloud -only and period.
It's not Illuminati who want to rule the world, it's Yakudza )))
Okay, jokes aside. Of course i checked out https://www.daikin.eu/en_us/products/product.html/BRP069C47.html . From drawings in the manual i couldn't even understand what it looks like. If it's a box or just a PCB. I could only realize it integrates much tighter and Wi-Fi credentials can be set up by a conditioner 's remote. @revk would be neat for Faikin too

Studying the PCB would give us knowledge of how this thing is powered. Additionally i am getting an interesting idea of showing Daikin a middle finger and porting Faikin to their original hardware. Because after all we've already paid for it. But for that i need to know what the core chip is and how it's interfaced.

⁣ King regards 
​

Sonic-Amiga, 2024-08-17:
By the way, Daikin manual labels the connector "S430", not "S21", and you know what that means ? S430 is NON-ISOLATED PORT!!! It has full AC main potential between any pin and earth!!! So be uber-careful messing with those, don't touch any running equipment by hands!

revk, 2024-08-17:
No clue how wifi could be integrated with remote!

yufaxt, 2024-08-17:
note that there was a report of successfully using pin 1 when 4 was missing 12v here https://github.com/revk/ESP32-Faikin/issues/54

not sure why some units are having these issues. 

Note that regarding wifi credentials via the remote, I believe this is incorrect. with the Emura 3 remote one could enable/disable wifi or activate WPS. For entering credentials one would need to connect to the Emura own Wifi network using the Onceta app and enter the credentials via the app.

I do however think there is place for some better integration with the remote. e.g not sure how the current behavior is, but would be nice if faikin notices external (hence not controlled by the auto mode) temperature change, and it’s running on auto mode, to use the temperature set in the remote to be the max (if cooling) or min (if heating). 
The use case here that I’m the “advanced” user at home, and use HA/Faikin web interface, but my better half would usually use the remote. If turned on via remote my pre-set faikin auto mode will kick in, but I think temp change will be only working via HA or Faikin web. Or is this already accounted for? Sorry for the offtopic

revk, 2024-08-17:
I have wondered about that, as we can detect an "external" change. We could use that to impact the setting - the issue is remotes don't send "up" and "down", they change what they have displayed and send that. One argument is any external change is a request to change the set point on Faikin Auto mode. That could work, even if it actually promptly sets it back way colder or hotter.

yufaxt, 2024-08-17:
This logic makes sense, in the end the direction is not important, the user has seen and “confirmed” the new temperature on the remote, and would expect the unit to respect it somehow. if the target temperature is set to that it should give the expected result. I would anyway keep the settings on +/-0.5 so the max/min temp is close to what is set in HA which reflects in the target temp, so if the remote set temp is reflected in the target temp the “user” will get an expected behavior.
For me personally this works because we only cool. Otherwise since usually one heats to reach lower temperature than one cools to, the +/-0.5 approach would mean to set the temp differently when changing the mode.
The other option would be (perhaps configurable?) to use the HA/remote set temp to set the max or min if cooling or heating respectively, instead of setting the target temp. that sounds a bit more complicated to implement though

revk, 2024-08-17:
I will consider as an option, probably on a setting anyway.

revk, 2024-08-18:
OK that idea is thwarted.

We *do* see when there is a change we did not ask for.

But if there is a change when we did ask, we don't see it, we keep asking many times until it is what we want.

I could change that, and record the old value we saw separately from what we are trying to set, but that still leaves a race where we are asking for a change *and* the user makes a change, and we see only one, maybe the one we are looking for.

The issue is that makes it unreliable to use the remote - some times remote makes a change, we make a change, we see ours confirmed. The problem is the cycle of polling is several seconds, making this quite easy to hit. My tests showed it is easy.

There is no feedback to the user - they set a new temp on remote and see on remote, but system does not see it.

I think doing this would be worse than not.

yufaxt, 2024-08-18:
Interesting. In this case I agree, better not complicate it. one can tell the user to use HA 😉 thanks for checking!

Take-Off90, 2024-08-19:
I started another test today and found out that the S21 port is the other way round. Shame on me... 🙄 

Faikin is now powered by my AC correctly and I can access the WebInterface.
Nevertheless Faikin tells me "System is offline"

I tried with and without connected PIN1 (5V)

Any idea what I am doing wrong?

Sonic-Amiga, 2024-08-19:
Perform a loopback test. You could have blown tx or rx on Faikin by supplying 14V there as result of reversal

⁣ King regards 
​

Take-Off90, 2024-08-29:
Thank you all for your help. 

It seems that the S21 port on Emura3 is the other way round. 
Faikin is now powered by the AC and connected. Most of the functions are working as expected. 

So I can confirm the following models as working:
- FTXJ35A2V1BB
- FTXJ25A2V1BB

I think the following models are working too as they are just different colour and size. (But not checked) I hope this helps if somebody is searching for Faikin + model code instead of "Emura3"
FTXJ20A2V1BW
FTXJ25A2V1BW
FTXJ35A2V1BW
FTXJ42A2V1BW
FTXJ50A2V1BW
FTXJ20A2V1BS
FTXJ25A2V1BS
FTXJ35A2V1BS
FTXJ42A2V1BS
FTXJ50A2V1BS
FTXJ20A2V1BB
FTXJ42A2V1BB
FTXJ50A2V1BB

Maybe you can also add the following link to your tutorials. For me it was very difficult to find the correct cable. On RS (Germany) you can only buy a big amount of cables. On AliExpress they are a lot cheaper and you are flexible in case your pin assignment is different.
https://www.aliexpress.com/item/1005005562174022.html?spm=a2g0o.order_list.order_list_main.11.4f801802nY81aZ

yufaxt, 2024-08-29:
Isn’t the direction determined by the plug‘s notch that is only fitting in one way? I found everything to work pretty much out of the box (crimped a 1:1 cable similar to the one daikin sells)

closed 2024-08-29 by Take-Off90

dahadd, 2024-12-05:
> Thank you all for your help.
>
> It seems that the S21 port on Emura3 is the other way round. Faikin is now powered by the AC and connected. Most of the functions are working as expected.
>
> So I can [...]

Can you help me please.
How do I connect the Faikin to the S21 port? Can you send me an assignment please . 
I also have an Emura 3 and there is no voltage on pin 4.

Pin 1 - 5V
Pin 2 - 5V
Pin 3 - 5V
Pin 4 - 0V
Pin 5 - GND

Thank you very much.

[image]

revk, 2024-12-05:
Are you sure you have pin 1 right?

dahadd, 2024-12-05:
hey revk :-)
I only get a voltages when I measure via pin5

Take-Off90, 2024-12-05:
Hello dahadd,

I can't open my unit right now, so I can only help you from what I remember.

Your picture looks correct. You should always measure against a known potential/ground.
Search for a known ground e.g. earth and measure each PIN against it.

Alternative you can measure each PIN against PIN1 and each PIN against PIN5. Be careful with polarity. 

I hope that helps.

u20p17, 2024-12-20:
> 
> @Take-Off90 Can you help me please. How do I connect the Faikin to the S21 port? Can you send me an assignment please . I also have an Emura 3 and there is no voltage on pin 4.
> 
> Pin 1 - 5V Pin 2 - 5V Pin 3 - 5V Pin 4 - 0V Pin 5 - GND
> 
> Thank you very much.
> 
> ![pinbelegung](https://private-user-images.githubusercontent.com/124442119/392840173-c37ca0d3-6d91-4531-97b8-c69f3c91143f.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2ODQ5NzMsIm5iZiI6MTczNDY4NDY3MywicGF0aCI6Ii8xMjQ0NDIxMTkvMzkyODQwMTczLWMzN2NhMGQzLTZkOTEtNDUzMS05N2I4LWM2OWYzYzkxMTQzZi5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMFQwODUxMTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNjJlZTExYjA2YmZkNDcxYTgwNWMyYTI2YmRkODdiY2E0YWQ5NDViNWFmY2U3MmQyY2NhN2M4NTgzYjhkZjc5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hjTn7nRJ-DrI-uDZ_NrN2lXe6_Ji7eT683hPFDGH5DI)

Hello! 
I have exactly the same issue. @dahadd: could you fix the issue?

I made an adapter cable from the S21 connector to the Faikin connector and had to swap the Pin 1 + Pin 5. I measure now the 5V correctly (I compared the measurements with the text which is printed on the pcb), but I do not measure 12V on pin4... @revk: will the faikin get powered via the 5V? Is there a reverse polarity protection on the board? I can not see a wlan hotspot from the faikin nor any light on the module...

revk, 2024-12-20:
The Faikin can run from the 5V supply (pin 1) which you would connect to pin 4 on the Faikin, *IF* there is enough current. So worth a try perhaps..

u20p17, 2024-12-20:
@revk: ok, just to understand it better: it seems that normally the faikin is power with the 12V, but as I do not measre any voltage there I should try to connect the 5V from Pin 1 to the 5V of the Faikin + additionally to the 12V of the Faikin?

revk, 2024-12-20:
I have no idea on air-con - it should have a triangle printed next to pin 1.

u20p17, 2024-12-20:
ok, then pin 1 is on the right... i just tried to wire the 5V directly to the 12V pin of the Faikin and it seems to work! (at least I see not the wifi hotspot.

Thanks!

dahadd, 2024-12-20:
@u20p17 which model do you have? Emura 3 FTXJ?

revk, 2024-12-20:
Need to check pins 2 and 3 are properly connected.

u20p17, 2024-12-20:
> @u20p17 which model do you have? Emura 3 FTXJ?

FTXJ35A2V1BW

> Need to check pins 2 and 3 are properly connected.

i can not find any issue there. could it be related that at the moment pin1 of the faikin is not connected to 5V?

revk, 2024-12-20:
That is possible, and if so, you need to connect the 5V to pin 1 (as well as pin 4). Worth a try.

u20p17, 2024-12-20:
> That is possible, and if so, you need to connect the 5V to pin 1 (as well as pin 4). Worth a try.

same issue - i only get a timeout. the led of the faikin is flashing violett - maybe this indicates someting?

revk, 2024-12-20:
I think it indicates it is not communicating.

u20p17, 2024-12-20:
should i messure a voltage between gnd and pin 2 or pin3 ?

revk, 2024-12-20:
Both should show around 5V.

dahadd, 2024-12-20:
Yes Pin 2 and 3 = 5V

revk, 2024-12-20:
OK I am assuming it should be S21, so you can set the `no` settings for other protocols (`nox50a`), and set the `noswaptx` and `noswaprx` to lock it all to S21.

Then maybe power cycle everything including the air-con.

u20p17, 2024-12-20:
ok, i can confirm that I measure 5V on Pin2 and 3... i also tried to set all no settings and made a power cycle but sadly i still get the timeout

revk, 2024-12-20:
OK not sure what to suggest.

u20p17, 2024-12-20:
do I have to set the GPIO pins? just saw that the Tx pin is set to -48 and the Rx to -34 in the webinterface

revk, 2024-12-20:
Those are right (`-` means inverted, which they are).

Short of you having 2 and 3 swapped by mistake or something, only real thought is faulty board.

u20p17, 2024-12-20:
i checked the wiring now again, but it is fine - as a test I tried to change the Tx and Rx pins in the webgui but also didn't help... so for me it also seems that the faikin board might be faulty :(

revk, 2024-12-20:
In that case, return for refund, sorry about that. It is rare.

u20p17, 2024-12-20:
no problem - i asked now for refund on amazon. nevertheless, I will be on business trip in UK on 06.01. and will send it back when I am there... maybe i can order another faikin to the hotel...

revk, 2024-12-20:
Sorry about that. We do some checks on every board anyway, but there can be some faults that slip through.

u20p17, 2025-01-12:
just a small update: i am back from my business trip and have bought another pcb - it is now working as expected 🎉 thanks!