revk/xESP32-Faikout #436: [BUG] MQTT topic names changed new version

klausj1, 2024-08-14:
**Faikin hardware**
Bought from Amazon

**Daikin hardware**
S21

**Describe the bug**
With the version Faikin-S3-MINI-N4-R2: 4cc882f 2024-06-10T16:02:39, the topic names for temperature and humidity are `bletemp` and `blehumidity`.
With the new version (auto-updated 4 hours ago) Faikin-S3-MINI-N4-R2: b16bfc4 2024-08-12T14:02:04, these names changed to `ble_temp `and `ble_humidity`.

(I have connected a temp sensor via BLE)

So my HA configuration is not compatible anymore.

Do I have to change my UI and my scripts in HA or is there another option?

**To Reproduce**
Just compare the two versions

**Expected behavior**
Being compatible wood be fine.

revk, 2024-08-14:
Wow, ok, any field changes should involve updated HA config being sent, so I'll investigate that.

revk, 2024-08-14:
I have just checked the code.

The HA config sets `bletemp` and `blehum` and `blebat` which is what is reported in the `state/name` message `HA` should be monitoring.

The is, as before, a `state/name/status` periodic message that reports a `ble` object containing `temp`,`hum`, `bat`, and `volt`.

I cannot find `ble_temp` or `blue_humidity` anywhere.

Do you have details of the MQTT messages you are seeing?

eupesco, 2024-08-14:
I have the same behaviour - HA names them with an underscore. MQTT messages are the same - bletemp etc.
May be related to the latest HA updates.

revk, 2024-08-14:
OK mqtt messages are the same, so where do we see the HA names?

Is this in the MQTT HA config we send?

revk, 2024-08-14:
We send a `name` of `BLE Humidity` I think, is HA using that somehow?

eupesco, 2024-08-14:
Yes, could be. In `homeassistant/sensor/****bletemp/config` , there is `name: BLE Temp`.

revk, 2024-08-14:
OK did not realised we had changed name, but surely it doesn't break anything?

klausj1, 2024-08-14:
This is what I see in the MQTT explorer:
[image]

The two topics there as text (I choose the humidity):

This is from the Faikin with the old version:
`{"unique_id":"DC5475EFC190blehum","dev":{"ids":["DC5475EFC190"],"name":"faikinSz","sw":"4cc882f","mf":"RevK","cu":"http://faikinSz.local/"},"icon":"mdi:water-percent","name":"blehum","dev_cla":"humidity","state_class":"measurement","stat_t":"state/faikinSz/ha","unit_of_meas":"%","val_tpl":"{{value_json.blehum}}"}
`
HA creates this name out of it: `sensor.faikinsz_blehum`

And this from the Faikin with the new version:
`{"unique_id":"F412FA97DB18blehum","dev":{"ids":["F412FA97DB18"],"name":"faikinAz","sw":"b16bfc4","mf":"RevK","cu":"http://faikinAz.local/"},"icon":"mdi:water-percent","avty_t":"state/faikinAz","avty_tpl":"{{value_json.up}}","pl_avail":true,"pl_not_avail":false,"name":"BLE Humidity","dev_cla":"humidity","state_class":"measurement","stat_t":"state/faikinAz","unit_of_meas":"%","val_tpl":"{{value_json.blehum}}"}`

HA creates this name out of it: `sensor.faikinaz_ble_humidity`

Don't these topics come from the Faikin?

revk, 2024-08-15:
Well yes, but as you quote, the topic is "avty_t":"state/faikinAz", for example.

Ie activity_topic is state/faikinAz

HA is making up a sensor name. If there is a way we can control that better with some field we send in that config, let me know.

klausj1, 2024-08-15:
Probably I should have told that I am using two faikins, one with the old Faikin version (faikinsz), and one with the new version (faikinAz).

HA makes the sensor name based on the topics I posted above - at least the last part.

The old version of the Faikin configures it like this: `"name":"blehum"`
And the new one: `"name":"BLE Humidity"`

And thats why the entity names in HA are incompatible. `sensor.faikinaz_blehum` changed to `sensor.faikinaz_ble_humidity`.

I only mean the last part of the name. faikinaz vs faikinsz is because of two different devices, thats totally fine.

The problem is not the state/faikinAz-topic, this contains the values. The problem is the "config" topic under homeassistant/sensor, because they define the names in HA.

revk, 2024-08-15:
I did not know HA did that. I assume it just works though generally. Is it other automations you have that refer to it by name, that break?

klausj1, 2024-08-15:
The problem did not happen with an HA update. The problem happened with a Faikin update. There are no other HA automations with this problem.

The problem was introduced with [this commit](https://github.com/revk/ESP32-Faikin/commit/df1d8b1ccbf6c0bf5e181b26be0722c14111673b), rows 2475, 2484, 2516.

revk, 2024-08-15:
Yeh, I gather than, I did not know the "name" was any more than cosmetic.

klausj1, 2024-08-15:
HA as a somehow complex ID and naming concept ...

Will you change this back, or shall I adapt my HA configuration?

revk, 2024-08-15:
Well, if I change it back, someone else, that has just started using it this way, will have the same problem. Best thing is for me to be a lot more careful changing "name" in future. Sorry.

revk, 2024-08-15:
And thanks for pointing it out.

closed 2024-08-15 by revk