revk/xESP32-Faikout #441: [BUG] 'Powerful' mode doesn't turn on

Sonic-Amiga, 2024-08-17:
**Faikin hardware**
Self-built using an off-the shelf ESP32 board and self-designed adapter

**Daikin hardware**
ATX20K2V1B S21

**Describe the bug**
I turn on "Powerful" mode in Faikin web UI, nothing happens, after some time the switch falls back to Off
When i turn Powerful mode from remote; it's correctly reflected in UI

The bug was discovered while testing the protocol. On my A/C F3 response byte[3] does not reflect powerful state; only F6 byte[0]

Sonic-Amiga, 2024-08-17:
Just out of curiocity tested "Comfort" function, which it also has. Faikin doesn't even detect it in the UI.

Sonic-Amiga, 2024-08-17:
Connected my BRP controller back to this unit and checked what happens in get_control_info and get_sensor_info when i select these special modes.

"powerful" is signalled by adv=2 in get_control_info
Other two don't change anything at all. Perhaps they are RC-only ???

I can see f_rate=A when Comfort mode is selected, but that's Auto fan speed, and yes, it resets fan to Auto.

revk, 2024-08-18:
OK, we had logic for using F3 or F6, which may have been changed unintentionally in latest beta. I'll have a look now.

revk, 2024-08-18:
Can you use `dump` to see what we are sending?

Logic should be if we get `F3` response we use `D3`, and if we get `F6` response we use `D6`.

Sonic-Amiga, 2024-08-18:
I will check, later. Having a rest today, tired of sitting in front of a display, sorry.
But your logic doesn't work on my unit. It responds to both F3 and F6, and F3 response is always 30 30 30 00
Interesting fact is that my BRP controller starts NAKing the response, containing 02 in the last byte.

It's a real pity you never live any comments in your code. What model have you seen using F3 byte[2] for 'powerful' state ?

Later i'll try changing the logic to inverse: "if we have F6, use D6; if we have F3, use D3"

revk, 2024-08-18:
I do have some comments :-)

OK, I can certainly make it use one or the other. I don't always know the models of units.

If there is a separate model/flag saying which to use, from the S21 work, that would be a better criteria for deciding.

Sonic-Amiga, 2024-08-18:
Yes, but you see, so far i couldn't decode the whole thing; and currently they are showing me a middle finger :) I will beat that eventually, but it appears harder than expected

Sonic-Amiga, 2024-08-18:
Commit 0c5f769894365d344012b36846f0b91023840d94 - what model what is made for ? Would be nice to reach the owner and ask to get replies for F2, F4, F6, F8, FK

If the unit doesn't reply to F6, that would be the easy way.

revk, 2024-08-18:
Yeh, even comments would not say that, short of trawling all of the issues raised on here, I would not know.

As you say, we can make it a "one or the other" if it does not reply on one.

ayasystems, 2024-08-19:
I've the same issue with a FTXS50G 

If i enable powefull it going off after few seconds

Faikin use presence sensor? I cann't view any option in webpage

Sonic-Amiga, 2024-08-19:
Hello everyone! Working on a fix.

Sonic-Amiga, 2024-08-19:
> "blame" may track when the F3/F6 change was made

That's how i found commit hash, with proud "checkpoint" in the message and zero references. We don't even know which model uses D3/F3

Sonic-Amiga, 2024-08-19:
... "powerful in:body" didn't help either. Okay, i see that change and i hope i understand the logic.

Sonic-Amiga, 2024-08-19:
... But i found a dupe of this ticket: https://github.com/revk/ESP32-Faikin/issues/215

Sonic-Amiga, 2024-08-19:
Hmmm, D6 command doesn't work either. Will try to resort to help of my BRP. Not sure if it has respective controls, but...

Sonic-Amiga, 2024-08-19:
Unfortunately BRP doesn't want to tell me anything. Attempt to turn on powerful mode by request:

http://192.168.0.104/aircon/set_special_mode?set_spmode=1&spmode_kind=1

gives me "ret=PARAM NG"

Apps are also greedy. Neither Onecta nor old app give me full controls; only basic ones: temperature, mode, fan speed, swing. Perhaps Daikin declared my controller outdated and only provides rudimentary support :(

Sonic-Amiga, 2024-08-19:
I made a small [enhancement](https://github.com/revk/ESP32-Faikin/pull/445), so that i see all responses. My ATX20K2V1B does not respond to D6, and D3 does nothing, at least with regard to "powerful" mode

{"protocol":"S21","cmd":"D3","payload":"32303030","text":"2000","ack":true}
{"protocol":"S21","cmd":"D6","payload":"30303032","text":"0002","nak":true}

Sonic-Amiga, 2024-08-19:
https://github.com/revk/ESP32-Faikin/pull/446 makes the setting work on FTXF20. ATX20 displays state correctly, if set by RC, but cannot control.

revk, 2024-08-20:
OK, OK, I'll fix "checkpoint" in future!

revk, 2024-08-20:
Try latest beta please.

closed 2024-08-20 by revk

ayasystems, 2024-08-20:
with my FTXS50G the same behavior. When I activate powerfull after a few seconds it is deactivated on the web

Faikin-S3-MINI-N4-R2: 35adf96 2024-08-20T13:57:47 S21

Sonic-Amiga, 2024-08-21:
Welcome to the club! I have more questions:

1. Does Faikin UI track "powerful" status correctly when controlled by remote?
2. What other special modes does your AC have? Mine has Eco and Comfort
3. Does UI track those other modes when set by remote?

On my ATX20 answers to (1) is yes; and to (3) is no

Sorry, there's no immediate solution available. Perhaps the solution doesn't even exist because i have original Daikin BRP online controller for this unit, and it also does not provide those controls.

I will eventually try brute-forcing to discover at least status poll requests, but no guarantees 

⁣ King regards 
​

ayasystems, 2024-08-26:
1. No, if i set powerfull with remote IR faikin is not showing as ON, still off

2. Mine have eco, comfort and moving sensor

3. No, it is not showing the button

Here the RX data 

[image]

Regards @Sonic-Amiga

dzungpv, 2024-09-06:
I have FTHF35 and it can only display Powerful mode, could not control with D3 or D6, Econo not display when using RC and could not control too, this should keep open until we find a solution @revk

Sonic-Amiga, 2024-09-06:
The same here with ATX20K2V1B. Apparently it uses different commands for special modes; or maybe even does not have them at all. I have original Daikin BRP069 adapter (two of them), and it also does not provide those controls on this A/C.

I am currently working on a heavy reverse engineering; but no idea how much time in total it would take.

Ideally to solve this we need to find any controller, which gives these functions on your A/C. We have quite advanced A/C simulator; we could plug that controller there and see what it does.