revk/xESP32-Faikout #449: [BUG] Gpio to GND without rezistor?

iulianhlc, 2024-08-22:
Hello,

I was looking at the schematic and I see a lot of GPIO pins to GND without a resistor. Is that ok?

revk, 2024-08-22:
Why would that be an issue? They start as either unconnected or input.

closed 2024-08-22 by revk

iulianhlc, 2024-08-22:
reason: not all gpio pins have pull-up/pull-down resistors

revk, 2024-08-22:
Sorry, not sure of the issue.

Leaving floating would be possibly bad if no pull up/down, but tied to GND would be fine.

Pins tied to GND should be fine unless pin starts as an output and outputs high. None of those do.

Tied to GND with internal pull up wastes some current, but the code actually checks all ports for that and ensures that if tied to GND it is set to a pull down instead.