revk/xESP32-Faikout #451: [FEATURE] manual setup of url (or dns suffix)

thib3113, 2024-08-24:
**Daikin hardware**
S403 (but I think it's not linked)

**Is your feature request related to a problem? Please describe.**
A problem but not really . In mqtt homeassistant config, it will add an url <name>.local . But, on my network, to access this I need to use <name>.<lastname>.lan . 

**Describe the solution you'd like**
Is it possible to have an advanced setting to set this url manually ?

revk, 2024-08-24:
Sorry, not sure I understand.

.local is normally MDNS stuff, and I am not sure the ESP IDf stack supports that by default.

But what do you want to add? The MQTT config has a host, and expects the host to resolve using DNS.

*OR* You can put an IP address directly in there. That will work.

thib3113, 2024-08-24:
In my network setup, if I go to .local . it will not resolve . 

And because this url is set up in mqtt : `/homeassistant/climate/<id>/config/dev/cu` . home assistant will create a button "visit" ... But if I click on it, I will end on a ERR_NAME_NOT_RESOLVED page . 

So, If it's "easy" to add an advanced setting, to force this url, I'm interessted . 

(thanks for really fast answers)

revk, 2024-08-24:
It should as we do have MDNS announced for .local

I may be able to look in to that, but how would it resolve to some other local host exactly?

thib3113, 2024-08-24:
my router do it . 

It get the name I set, or the hostname send . And create a local dns record : `<name>.thib3113.lan`

revk, 2024-08-24:
Surprised a .local does not also just work.

I'll think about it.

thib3113, 2024-08-24:
It's maybe because my IOT is in a special VLAN . My computers are not in the same vlan . So, mdns maybe doesn't pass through the firewall .

sm-Fifteen, 2024-08-27:
> Surprised a .local does not also just work.
> 
> I'll think about it.

It may not work under certain circustances, yeah. Besides the split vLAN situation OP is dealing with, another one I've encountered is how, on Linux, Flatpak-sandboxed applications have [no mean of performing mDNS resolution](https://github.com/flatpak/xdg-desktop-portal/issues/505), for instance. If your web browser is installed like this, you may need to use the IP address directly.

> So, If it's "easy" to add an advanced setting, to force this url, I'm interessted .

If mDNS is not an option, you can either figure out how to add a local DNS entry in your router's DNS tables (not all routers can) or [manually configure a mapping between a hostname and an IP address](https://www.hostinger.com/tutorials/how-to-edit-hosts-file) in the correct config file for your OS. It's usually reccomended to use [.home.arpa](https://www.rfc-editor.org/rfc/rfc8375.html) as the TLD for LAN-only domains, so DNS resolvers don't try to forward it to actual DNS servers.

revk, 2024-08-28:
OK new field in latest beta.

closed 2024-08-28 by revk

thib3113, 2024-08-28:
Tested, it works, thank you

commit 0e2dad49 (2024-08-28): Added ha.domain for links on HA (Fix #451)