revk/xESP32-Faikout #452: [FEATURE] proxy settings

thib3113, 2024-08-24:
**Daikin hardware**
Actual board on amazon / S403 (but I think it's not linked)

**Is your feature request related to a problem? Please describe.**
Actually, my faikin is in my IOT vlan ... and can't access internet directly . I search how to setup a proxi, but doesn't find

**Describe the solution you'd like**
A way to set an http proxy

**Describe alternatives you've considered**
a way to download firmware, and upload it to the faikin

revk, 2024-08-24:
Sorry, confused.

A proxy would be something you set to access the Faikin?

The only time Faikin tries to access http is s/w updates. You want it to use a proxy for that?

thib3113, 2024-08-24:
no, a proxy to allow faikin to access update server . 

In fact, I doesn't exactly know how to do, because I can't find a way to get the last firmware, nor a way to upload it to the faikin . 

---

I've done some search, and maybe I can try to reverse the update url, change the otahost, and create a custom app that will proxy the request, but not sure if it's a good idea too .

revk, 2024-08-24:
You can, indeed, set the hostname for update, and if that is a proxy, that will do it.

closed 2024-08-24 by revk