revk/xESP32-Faikout #454: Anohter No voltage case on S21 header of   FTX60GV1B

yiannosi, 2024-08-25:
My indoor unit above has a clearly marked S21 (see photos) but there is no voltage present neither on pin 4  or Pin 1 !

At first I thought that there was something wrong with the connector/ wiring but I also tried with standard dupont and it didn't make any difference.

[image]
[image]

Anybody came across this unit before ?  Any ideas ?

Sonic-Amiga, 2024-08-25:
First of all, read [this](https://github.com/revk/ESP32-Faikin/issues/434) and triple-check.

Hmmm... I see "+12V" mark on the PCB nearby...

sm-Fifteen, 2024-08-26:
It's possible you've just plugged it in reverse. I made that mistake when installing mine (there's really no good way to tell the correct orientation on that thing), and flipping it solved the issue with (luckily) no damage to any component that I could tell. I notice 2 capacitors right next to your S21 port, aligned with two of the pins, so my guess would be that these are the data pins.

But from your OP, I take it you've already tried both or tested the pins with a multimeter?

yiannosi, 2024-08-26:
> But from your OP, I take it you've already tried both or tested the pins with a multimeter?

I tested the pins with multimeter taking into account the possibility of different pinout. I just wish there was a known solid ground to use as a reference on the PCB. I had to partly dismantle the pcb from the rest of the AC unit while standing on a ladder and its not that easy to take measurements with the power switch on...

revk, 2024-08-26:
The little triangle denotes pin 1, if that helps.

yiannosi, 2024-08-28:
> Oookay, let's assume that we've already checked against stupid things and here go smart ones... Unfortunately that would require more ladder-up-n-down
> 
> From markings on the board we can see that GND is JP36 and +12V is JP35. So, we can beep which pin on S21 is ground. Pins next to C65 and C66 indeed look like tx/rx. So, if we line this up with the pinout we know, we'll get: ![image](https://private-user-images.githubusercontent.com/48695031/361533856-ca480347-74a4-44a4-92fd-c5a8c5105dc4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ4NzQwNTcsIm5iZiI6MTcyNDg3Mzc1NywicGF0aCI6Ii80ODY5NTAzMS8zNjE1MzM4NTYtY2E0ODAzNDctNzRhNC00NGE0LTkyZmQtYzVhOGM1MTA1ZGM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI4VDE5MzU1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVhM2M2YzEyYzE0Y2MxYjM3NTdmODE4MmYyZGFmYjk5M2YzNWRkNDgxYzNkODU2YjZlMGQzZDRkNDY3Mjg4ZDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MWRe7maqlagNRKluZ0yPDK-IMqjD2Pfs5OiYMGvjpYc)
> 
> It's now your turn to confirm a GND pin (or find a proper one) and check whether 12V is present on JP35. Further actions depend on whether 12V is present

Will do over the weekend

yiannosi, 2024-09-22:
Solved. Faulty connector issue.
Successfully installed on FX60GV1B

closed 2024-09-22 by yiannosi