revk/xESP32-Faikout #459: [BUG] Cant connect to wifi

Zappelphilipp, 2024-08-27:
**Faikin hardware**
Amazon 07.2024

**Daikin hardware**
S21 Port on Stylish

**Describe the bug**
Device shows as AP, i can connect to the Portal/Web UI but after setting the WIfi and pressing save nothing happens.

**To Reproduce**
Connect to Faikin AP, setup Wifi, Faikin does not connect to wifi

**Expected behavior**
Faikin connects to wifi and gets DHCP adress

**Additional context**
I have 2 wifis with about 75 iot devices without any problems. I also tried an android hotspot but same result. I read about a bug were the Wifi PSK needs to be shorter then 32 characters which is not the case on my wifis.

Zappelphilipp, 2024-08-27:
UPDATE: Ok, i really don't know why, but after the ~40 try it suddenly worked without changing anything. can be closed i guess :)

closed 2024-08-27 by Zappelphilipp

revk, 2024-08-28:
There was a bug, and a new faikin may have the older code until connected to WiFi enough to upgrade, perhaps.