revk/xESP32-Faikout #462: [FEATURE] Stylish Visor Control

Zappelphilipp, 2024-08-29:
**Daikin hardware**
Daikin Stylish (FTXA35C2V1BW9) on S21 Port

**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
The Stylish has a visor which opens before operation. In the daikin app there is a switch to manually open or close it which is quite handy for cleaning purposes.

It would be awsome if this feature could be implemented into the faikin firmware.

revk, 2024-08-31:
This is likely to come out of the reverse engineering project that some people are working on. If the app has it, then it should be possible to add at some point.

Sonic-Amiga, 2024-09-23:
Hello! You may assist reverse engineering.
1. Please run this script on your unit: https://github.com/revk/ESP32-Faikin/blob/main/Tools/Simulators/collect_profile.py and post the output
2. Do you have original Daikin controller which exposes this function ?

joealden, 2024-11-21:
@Zappelphilipp sorry to hijack this issue, but is there somewhere + would you be willing to answer some questions I have about the Stylish units (both with and without Faikin usage)? thanks!

Zappelphilipp, 2025-06-29:
Hi guys,

sorry for the late response. I tried to run the script but i get

```
Traceback (most recent call last):
  File "/tmp/collect_profile.py", line 150, in <module>
    mqttc = mqtt.Client(mqtt.CallbackAPIVersion.VERSION2)
                        ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'paho.mqtt.client' has no attribute 'CallbackAPIVersion'
```