revk/xESP32-Faikout #464: [BUG]

jmsucasas, 2024-09-02:
Hi,

I am trying to build the firmware from github source.  I have cloned it with:  

git clone --recursive https://github.com/revk/ESP32-Faikin.git

I have downloaded and installed esp-idf version 5.2.2 (same fail with 5.3),  and the following error appears when i try to set-target:  idf.py set target esp32s3

-- Configuring done
CMake Error at /home/jmsucasas/esp/esp-idf/tools/cmake/component.cmake:479 (add_library):
  Cannot find source file:

    /home/jmsucasas/ESP32-Faikin/ESP/settings.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  /home/jmsucasas/esp/esp-idf/tools/cmake/component.cmake:652 (idf_component_register)
  main/CMakeLists.txt:3 (register_component)

CMake Error at /home/jmsucasas/esp/esp-idf/tools/cmake/component.cmake:479 (add_library):
  No SOURCES given to target: __idf_main
Call Stack (most recent call first):
  /home/jmsucasas/esp/esp-idf/tools/cmake/component.cmake:652 (idf_component_register)
  main/CMakeLists.txt:3 (register_component)

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1, output of the command is in the /home/jmsucasas/ESP32-Faikin/ESP/build/log/idf_py_stderr_output_5857 and /home/jmsucasas/ESP32-Faikin/ESP/build/log/idf_py_stdout_output_5857

revk, 2024-09-02:
Are you running make not cmake?

Make makes the settings stuff.

closed 2024-09-02 by revk

jmsucasas, 2024-09-02:
I am using cmake as it appears in sdkconfig.defaults:
CONFIG_IDF_CMAKE=y

revk, 2024-09-02:
Ok use make, it will do the idf.py based make in due course.

jmsucasas, 2024-09-02:
How can i do it? Every idf.py (idf.py menuconfig,  idf.py set target) fails with the same error

revk, 2024-09-02:
Type the command

make

Simple!!!

jmsucasas, 2024-09-02:
jmsucasas@E5430:~/ESP32-Faikin/ESP$ make
make -C components/ESP32-RevK revk_settings
make[1]: se entra en el directorio '/home/jmsucasas/ESP32-Faikin/ESP/components/ESP32-RevK'
gcc -O -o revk_settings revk_settings.c -g -Wall --std=gnu99 -lpopt
revk_settings.c:5:10: fatal error: popt.h: File does not exists
    5 | #include <popt.h>
      |          ^~~~~~~~
compilation terminated.
make[1]: *** [Makefile:5: revk_settings] Error 1
make[1]: se sale del directorio '/home/jmsucasas/ESP32-Faikin/ESP/components/ESP32-RevK'
make: *** [Makefile:43: components/ESP32-RevK/revk_settings] Error 2

revk, 2024-09-02:
OK progress, you need the `popt-dev` apt to provide `popt.h`

jmsucasas, 2024-09-02:
It works!!! Thank you!