revk/xESP32-Faikout #465: [BUG]

jmsucasas, 2024-09-03:
Hi,

   I am trying to flashing the code (builded previously by me on linux machine) on a ESP32-S3 DevKit C and i receive the following error:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759c0
0x403759c0: esp_restart_noos at /home/jmsucasas/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3810,len:0x48c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa4c
load:0x403cc700,len:0x26e4
entry 0x403c986c
E (408) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (408) cpu_start: Failed to init external RAM!

I have tried to change the PSRAM mode to octal but the error remains:

���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759bc
0x403759bc: esp_restart_noos at /home/jmsucasas/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3810,len:0x48c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa4c
load:0x403cc700,len:0x26e4
entry 0x403c986c
E (408) octal_psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (408) cpu_start: Failed to init external RAM!

I not ESP32-S3 DevKit C valid chip to test it?

Thank you in advance
BR

revk, 2024-09-03:
The `s3` build is for an ESP32-S3-MINI-N4R2 which has SPIRAM

Looks like you don't have SPIRAM?

You will need to do a special build, e.g. 
```
components/ESP32-RevK/setbuildsuffix -S3-MINI-N4
make
```

Or use an ESP32-S3-MINI-N4R2

jmsucasas, 2024-09-03:
Thank you!!  It works with another ESP32-S3 board

closed 2024-09-03 by jmsucasas