revk/xESP32-Faikout #466: Connect to mqtt broker with certificate

jmsucasas, 2024-09-03:
Hi,

   This project uses RevK libraries to connect to MQTT Broker.  Is it possible to connect with broker that needs ca.cert,  client.key and client.cert to connect with?

Thank you in advance,
BR

revk, 2024-09-04:
The library has CA cert, and client key/cert, settings but they are not on web config.

I have added to web config now. See if that works.

Beta issued in a minute.

jmsucasas, 2024-09-04:
Nothing appears on advanced settings tab:
[image]

revk, 2024-09-04:
Weird, and yes, I see the same here, so I'll have to take a look.

closed 2024-09-04 by revk

jmsucasas, 2024-09-04:
I have buit it again,  but is not possible to write any value in the textfields:
[image]
No WiFi,  no MQTT

revk, 2024-09-04:
Wait, is this your own build?

You'll need a `make update`

jmsucasas, 2024-09-04:
I have deleted all the sources and build it again

revk, 2024-09-04:
You just needed `make update` to get the new library I expect. Or even just `make pull` to pull new library in.

jmsucasas, 2024-09-04:
I have executed make pull:
jmsucasas@E5430:~/ESP32-Faikin/ESP$ make pull
git pull
Ya está actualizado.
git submodule update --recursive
Ruta de submódulo 'ESP32-BLE-Env/components/ESP32-RevK': check out realizado a '671574b53a3674fb0eacdbe6874b6818aa75cb8d'
Ruta de submódulo '../Tools/SQLlib/stringdecimal': check out realizado a '8f121f25da097e0673b292f038f7600df1f2eb45'
jmsucasas@E5430:~/ESP32-Faikin/ESP$ 

and i have the same problem.  I cannot fill information in checkboxes that appears in captive portal

revk, 2024-09-04:
OK, that is weird, as, well, it works fine for me!

What chipset?

revk, 2024-09-04:
OK I am using an S3 (N4R2), and not having a problem.

Have you tried flashing the binaries in the beta release folder?

jmsucasas, 2024-09-04:
I have tried to flash with esptool.py --chip esp32s3 write\_flash -z 0x1000  Faikin-S3-MINI-N4-R2.bin
but chip continuously restarts

jmsucasas, 2024-09-04:
I have accomplished it with:  esptool.py --chip esp32s3 -p /dev/ttyACM1 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x10000 Faikin-S3-MINI-N4-R2.bin  but problem remains.

revk, 2024-09-04:
Wait, using my binaries?

I am really puzzled now, as this is working for me, honest.

I'll have a look.

revk, 2024-09-04:
Tried on another board, and also all works.

Not sure what it could be?

jmsucasas, 2024-09-04:
I am trying with a S3 mini board,  and i receive the following error.  It does not start the captive portal:

Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Running idf_monitor in directory /home/jmsucasas/ESP32-Faikin/ESP
Executing "/home/jmsucasas/esp-r53/esp-idf/bin/python_env/idf5.3_py3.10_env/bin/python /home/jmsucasas/esp-r53/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /home/jmsucasas/ESP32-Faikin/ESP/build/Faikin.elf -m '/home/jmsucasas/esp-r53/esp-idf/bin/python_env/idf5.3_py3.10_env/bin/python' '/home/jmsucasas/esp-r53/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.4.0 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048ede
0x40048ede: recv_packet in ROM

SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce2810,len:0x48c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xa50
load:0x403cb700,len:0x26f8
entry 0x403c886c
E (62) task_wdt: esp_task_wdt_reset(705): task not found
**E (1079) httpd: httpd_server_init: error in listen (112)**

jmsucasas, 2024-09-04:
After lots of flashings an erasing,  it works on the first board.  But there is a problem:  If the broker needs certs to connect to,  the option to add a certificate does not appears it MQTT is not connected. 
Is needed to codify the certificates in base64?

revk, 2024-09-04:
The certs / keys are all base64

jmsucasas, 2024-09-05:
I have found the error on the captive webpage.  The error appears to be on maxlength=0:

[image]

Could you check it?

revk, 2024-09-05:
Oooh, OK, that is easy to fix then, weird if that killed the rest of the page somehow!

revk, 2024-09-05:
OK I see the same, but the code does not seem to make that possible, so this will be fun to find.

revk, 2024-09-05:
Found it, making a beta now.

revk, 2024-09-05:
Glad the inputs work.

textboxes will have no issue, but the http post handling may

But that will be a lot more work. I'll look in to it.

jmsucasas, 2024-09-06:
How can  I see the following traces?
         ESP_LOGE (TAG, "Connect MQTT%d %s as %s", client, config.hostname, config.client);
It appears not to be painted on the serial interface

revk, 2024-09-06:
Serial monitor.

jmsucasas, 2024-09-06:
Is it possible to store certificates in sdkconfig by executing "make menuconfig"?
This will solve the problem to send high amount of data along POST.

revk, 2024-09-06:
very possibly

jmsucasas, 2024-09-07:
Please help me.  How can I copy the CONFIG_REVK_MQTTCERT defined in Kconfig to mqtt cert variable in revk.c?

revk, 2024-09-07:
A build should make it the default.

jmsucasas, 2024-09-07:
Certificates in PEM format is something like:
-----BEGIN CERTIFICATE-----
MIIC+TCCAeGgAwIBAgIJAJKo4+ykJQ0PMA0GCSqGSIb3DQEBCwUAMBExDzANBgNV
BAMMBm15LXNpdGUwHhcNMTkxMTA5MDk0ODQ5WhcNMjAxMTA4MDk0ODQ5WjARMQ8w
DQYDVQQDDAZteS1zaXRlMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEA
...
-----END CERTIFICATE-----
I think that the library rejects to load it due to "-----BEGIN CERTIFICATE-----" and "-----END CERTIFICATE-----" is not codified in base64?  Could be possible?

revk, 2024-09-07:
Yes you just want that base64 content!