revk/xESP32-Faikout #471: AC powering up by itself at night

Cees-tech, 2024-09-05:
Hi, I'm using Faikin (with my own hardware) on a Daikin FTXM35R5V1B.
[image]

Since I installed it I noticed the AC going from idle to power on (cooling) randomly during the night. 
Faikin auto is not in use.

When looking at the Faikin web interface the Main Power switch is toggled on. It looks like Faikin sends a power on command during the night. I'm using MQTT / homebridge to control the AC with Homekit. But no power on command was send over MQTT. In Homekit the status of the AC is still powered off/idle.

{"ts":"2024-09-05T17:45:39Z","id":"F0F5BD6D68E8","up":true,"uptime":541341,"mqtt-up":36015}

{"protocol":"S21","online":true,"home":25.5,"heat":false,"fanrpm":530,"comp":14,"outside":26.5,"liquid":12.0,"anglev":81,"Wh":3318600,"power":true,"mode":"C","temp":25.0,"demand":100,"fan":"A","swingh":false,"swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":false,"quiet":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}

{"protocol":"S21","ts":"2024-09-05T18:04:02Z","online":1.00,"home":[25.50,25.50,25.50],"heat":false,"fanrpm":[530,530,530],"comp":[14,14,14],"outside":[26.50,26.50,26.50],"liquid":[12.00,12.05,12.50],"anglev":[81,81,81],"Wh":[3318600,3318600,3318600],"power":1.00,"mode":"C","temp":[25.00,25.00,25.00],"demand":[100,100,100],"fan":"A","swingh":false,"swingv":false,"econo":false,"powerful":false,"comfort":false,"streamer":false,"quiet":false}

Hopefully someone can help me with this. Please let me know what additional information, data or loggings are needed if you need more?
Thanks, Cees

revk, 2024-09-05:
That is a feature, but only for Faikin auto mode set, and I think a setting?

Cees-tech, 2024-09-08:
> That is a feature, but only for Faikin auto mode set, and I think a setting?

Where can I find this setting?

I've changed a few things in mqttthing regarding power on and off and until now no more random turning on during the night. So it looks like the problem is already solved.

revk, 2024-09-09:
Automatic power on is effective is `autop` is set.

Described as *Enable auto power on/off* in the settings.

closed 2024-09-09 by revk