revk/xESP32-Faikout #472: [Question] Check model S3 working or not.

hoangtridung, 2024-09-06:
In my place, I can only buy one of 3 model: ESP32-S3: N8R2, N8R8 or N16R8.
Is it possible work with this firmware?

revk, 2024-09-06:
The N8 build will work with those as it just won't use the SPIRAM

hoangtridung, 2024-09-06:
I flash the sample board (S3 N16R8 ) by command line:

_esptool.exe -p COM14 write_flash 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N8.bin_

Wiring to Daikin as:

**(+5V) → HV( Shifter)**
**Pin2 (TX) → HV1( Shifter) → LV1( Shifter) → GPIO44**
**Pin3 (RX) → HV2( Shifter) → LV2( Shifter) -->GPIO43**
**Pin4 (+14V) → DC-DC Regulator              → VCC**
**Pin5 (GND) → DC-DC Regulator               → GND**
**_______________________________ LV              <-- 3V3**

And I can control almost feature on my Daikin, except for the powerful mode and preset.  I test with Daikin model FTXS25EVMA and FTXS50EVMA. The minor problem: sometimes I cannot type anything to change the config on Web UI.

**Thank you for your effort !**
Do you plan to implement it on the C3 module in the near future?

Sonic-Amiga, 2024-09-12:
Missing powerful mode is a known problem of FTXS. It looks like Daikin doesn't support it either; so probably there's no solution.