revk/xESP32-Faikout #482: [FEATURE] Add an option to control LED on Faikin auto mode based on HVAC action

Hellslicer, 2024-09-17:
**Daikin hardware**
Perfera Floor Standing Unit (FVXMxxA3V1B); S21

**Is your feature request related to a problem? Please describe.**
With Faikin auto mode and Floor Warming mode, we don't have any visual feedback on the indoor unit of the current HVAC action (idle or heating in heat mode).

**Describe the solution you'd like**
If LED control is enabled (`no.led=0`), we can reflect the current HVAC action using the LED:
- HVAC action is `heating` or `cooling` => LED is on
- HVAC action is `idle` => LED is off

This new behaviour could be enabled with a new setting.

**Describe alternatives you've considered**
We can automate this behaviour within HA.

**Additional context**
##### New setting
```
bit	auto.lcontrol	0		.live=1					// Auto toggle on/off LED depending on current HVAC action
```

revk, 2024-09-17:
Do you want to do an MR?

Or it may take me a while to get around to it :-)

Hellslicer, 2024-09-17:
Ok I can try :)

Hellslicer, 2024-09-19:
How can I disable OTA verification to test my local build?

revk, 2024-09-19:
Build without it and flash a need bootloader as well using serial/USB.