revk/xESP32-Faikout #484: [BUG] HVACAction IDLE not reported to home assistant

MUN0X, 2024-09-17:
The HVACAction "Idle" is no longer reported to home assistant.
When faikin start heating or cooling in auto-mode these states are reported correctly. 
However unit will never go to "idle" (which messed with some of my automations)

When looking at the mqtt messages, the issue appears to be in the faikin logic as it keeps sending state messages with `action: "heating" or cooling` but never changes to idle. message changes to "off" when unit is turned off.

Hellslicer, 2024-09-18:
A fix is available on beta release (https://github.com/revk/ESP32-Faikin/pull/481)

MUN0X, 2024-09-18:
Reporing of hvac_action works fine on latest beta.
Thank you for the quick response!

closed 2024-09-18 by MUN0X