revk/xESP32-Faikout #485: [BUG] Issue OTA update with Beta switch on

urbozz, 2024-09-18:
**Faikin hardware**
Amazon purchased, '1st gen'

**Daikin hardware**
S21 FDXS50CVMB

**Describe the bug**
Cannot update latest beta. Error - Upgrade check failed

**To Reproduce**
Steps to reproduce the behavior:
1. Set beta switch on
2. Click 'Upgrade now from ota.revk.uk (beta)'
3. Get error 'Upgrade check failed'

**Expected behavior**
Upgrade to latest Beta

**Additional context**
Footer info: Daikin-S1-PICO: b16bfc4 2024-08-12T14:01:38 S21

[image]

revk, 2024-09-18:
Very old if app name is Daikin! Try now.

closed 2024-09-18 by revk

urbozz, 2024-09-18:
Works, thanks!