revk/xESP32-Faikout #488: [BUG] env is not reported properly in MQTT Faikin/ topic if BLE is active

miniluma, 2024-09-18:
running the latest beta from 17.09.2024

**Describe the bug**
`livestatus` active
`fixstatus` inactive
`reporting` = 300
`nofaikinauto` active

1) (as expected)
Faikins with environment monitoring data received via MQTT command report `env`, and the env array in `Faikin/ `
Info: I send the measured-temp from a Homematic room sensor with `command/$DEVICETOPIC/control {"env":"$EVTPART1"}` (with `$DEVICETOPIC` as the faikin name and `$EVENTPART1` as temperature value).

2) (unexpected)
Faikins with BLE send env in `status/` but `Faikin/` only receives the array. The current value is missing.
The ble_.* as part of the status is sent as expected and equals the env value.

revk, 2024-09-22:
Hmm, the `"temp"` report in status is the `env` we are using if set, else `home` if set, else `home` if set.

The `env` is set from BLE if available, or `command` with `"env"`, if being sent, so either of these will report the `"temp"` in status.

Separately, the status also includes `"bletemp"`, `"blehum"`, and `"blebat"` if available (though maybe they would be better in a `"ble`" object. Hmmm.

The `Faikin` message reports periodic statistics (array of min/ave/max, or single value, depending on setting) that are collected from the Daikin. But this includes `env` (set by BLE or `command`). It does not include the ble data separately.

So I think working as I designed...

miniluma, 2024-09-24:
ok, when I disable `livestatus`, I see it as part of `Faikin` as well.

With `livestatus=1`, I see the env array (min/avg/max) in Faikin, but `Faikin/env` is not included. (`status/env` is updated as expected)

revk, 2024-09-24:
OK that makes no sense.. I'll have to have another look.

But it was never in `Faikin` as it is not a *real* field from the aircon, the real fields are things like `home`, etc.

revk, 2024-09-24:
Depends which reporting, you can find `env =` in a few places, setting from temp from aircon, or BLE, or the `control` messages.

revk, 2024-09-24:
Sounds like the test is wrong there!

revk, 2024-09-24:
That means `fixstatus` was wrong all along, I am fixing now - beta in a moment.

miniluma, 2024-09-24:
tested with fixstatus == 1: Works perfect! Thank you. 

untested for fixstatus == 0.

closed 2024-09-29 by revk