revk/xESP32-Faikout #498: [FEATURE] Expose Dark mode

Hellslicer, 2024-09-23:
**Daikin hardware**
Perfera Floor Standing Unit (FVXMxxA3V1B); S21

**Is your feature request related to a problem? Please describe.**
Current Dark mode is not applied live and not exposed on HA.
It can be useful to add an automation with dark mode control, i.e. to enable dark mode during night hours.

**Describe the solution you'd like**
- `dark` setting with live flag to avoid a reboot on changes
-  Dark mode exposed (MQTT + HA switch) like many other settings

I can open a PR with proposed changes.

**Additional context**

```diff
-bit	dark								// Dark mode
+bit	dark			.live=1					// Dark mode
```

Sonic-Amiga, 2024-09-23:
Hello! So does your unit have LED control! Cool, i am working on full reverse engineering of Daikin protocol; and we're looking for some info.

1. If you have original Daikin controller with Onecta app; does Onecta provide this control ? If yes, could you make a screenshot of what it looks like ?
2. Could you run [this script](https://github.com/revk/ESP32-Faikin/blob/main/Tools/Simulators/collect_profile.py) on your unit and post results ? This script tries to collect identification data from your AC; as much as it can. It does no harm, does not change any settings and does not expose any private info. We're trying to figure out how to autodetect features.

Hellslicer, 2024-09-23:
Hello,

1. I didn't find LED control on Onecta app (with original Daikin controller connected on S801)
2. This is a bit off topic here, so I'll post the results on https://github.com/revk/ESP32-Faikin/issues/408

revk, 2024-09-24:
I'll make dark mode a live setting. Beta due shortly

MQTT allows setting changes, but I have not done any settings on HA switches as yet, so that is a separate thing to consider in future.

closed 2024-09-24 by revk