revk/xESP32-Faikout #506: [BUG] Cannot activate Faikin Auto Mode using MQTT publish

Taxxian, 2024-09-28:
**Faikin hardware**
Amazon

**Daikin hardware**
4 different Stylish Wall Device with S21

**Describe the bug**
Is sent various MQTT commands like:

command/FaikinWozi/control
{
  "env": 23.0,
  "target": [22.0,23.0]
}

command/FaikinWozi/control
{
  "target": 24.0,
  "env": 21.0
}

It is received and I can see the new "env" value in MQTT status.

But "target" never changes and "control" is always false.

I can activate Faikin Auto Mode and set "target" in the Web UI and I can then see this data in the MQTT status, but if I do it using MQTT publish "target" values never change and FAM never avtivates.

**Expected behavior**
I shoud see the new "target" values in the WEB UI and MQTT status, FAM Mode should activate.

**Additional context**
Everything I tested works fine, I updated to the new Faikin version and it still works fine, except for the "target" command. (Dont know if it worked before the update, own it for 3 days...)

Thanks for the help!

Regards

Martin

revk, 2024-09-29:
OK, this is exactly how I use it, and it works for me, honest.

e.g. I regularly send

```
command/OfficeAC/control {"env":21.50,"target":[21.000,21.500]}
```

And the `Faikin/OfficeAC` message includes `"control":true`

You need two send `control` regularly. The "Faikin auto mode" on the web interface should disappear, e.g. 

[image]

I'm running latest beta.

Taxxian, 2024-10-02:
Interesting I think I got it to work so far.

I can set the desired temp in the exposed climate entity in HA that directly translates to "target".
Then I had to activate lock mode to prevent the Faikin from going to cooling mode when night comes and the thermostat goes to 18Deg.

What does "update regularly" mean? Dont know if setting env alone keeps FAM running longterm,

Other stuff:
The climate entity shows heat/cool which is suboptimal, how can I see if its heating or not? I have to manually switch from heat to cool or back  2 times annually the way it is set up, and I have no solution for that.

Those icóns are incomprehensible, text should be standard.

Is there a way to influence the compressor frequency of the outdoor unit? I have a certain range where my roof vibrates, I want to skip that...

Overall: I like it! :-)

revk, 2024-10-02:
Arg, ok I think send like every minute or so.

Not sure easy to change the heat/cool state. People want to know it is auto. But I guess it "locked" it may be something that ca  be fixed.

Taxxian, 2024-11-21:
Update:
My workaround is doing fine for over a month now, so I dont consider this a problem anymore.