revk/xESP32-Faikout #511: Unresponsive faikin board

sywesk, 2024-10-01:
Hello,

While installing my faikin modules on my new Daikin emura 3 splits, one of them disappeared never to return. No led, no wifi. I checked the pins and had around 13 volts so that looked OK. 

I think this happened after I cut the mains for that multi-split unit before taking apart the next split.

I bought a new board to replace it, but do you think it is fried? Is it worth buying the tap cable to try to reprogram it?

I didn't use the "bug" template as this is most likely an end user issue. 

Thanks!

revk, 2024-10-01:
Should work up to 36V, but if there was some power spike of some sort, maybe.

You may be able to solder on to the pads to connect a USB to check if it responds.

sywesk, 2024-10-01:
Excellent idea, I'll try that tonight!

sywesk, 2024-10-01:
I soldered and connected the board to my laptop: it did show up and boot as expected. But after plugging it back to the AC, still nothing. I tested thoroughly the cable again, but it was fine. 

My guess: The 12v -> 5v converter is dead. To confirm this, I modified my cable to remove the 12v wire, and connect the unit's 5v directly to the tag header 5v pin, and it worked.

Do you have any idea what might have caused the voltage regulator to fry ? Maybe some electrostatic electricity while manipulating the board ?

revk, 2024-10-02:
There have been quite a few reports of 12V not working on Daikin units. In many units the 5V does not have enough current, but if working for you that is great news. Some do have enough on 5V.

closed 2024-10-02 by revk