revk/xESP32-Faikout #512: [BUG] Thermostat auto mode is not working fine

ayasystems, 2024-10-02:
**Faikin hardware**
 Faikin-S3-MINI-N4-R2: 58657f3 2024-09-30T08:20:19 S21 (Amazon)

**Daikin hardware**
Connected with S21
FTXS50G

**Describe the bug**
In auto mode with timer daikin turns on with the temperature below the target in heat mode.
I'm using BLE sensor and at right you can see a graph with the values of this sensor
 
[image]

[image]

Target is set as 21 and when time is working the room temp goes to 24,5 and remains at 24,5

revk, 2024-10-02:
The logic is pretty simple, so that seems rather odd. Not sure if anyone else has seen that.

May be worth setting thermostat setting as that makes the logic much simpler. There are a number of other settings that can be adjusted as well.

ayasystems, 2024-10-02:
Without external BLE sensor seems works fine. I will try this night to see the diferent

[image]

ayasystems, 2024-10-03:
Yesterday I set the thermostat without the external thermometer.
Tonight the temperature did not drop below 23º and the thermostat was set at 22.5. The machine has not turned on all night.
It seems that in my case it fails when I use an external BLE thermometer.

revk, 2024-10-03:
That really is odd, as lots of people are using this. I'll double check and see if I can see any cause.

ayasystems, 2024-10-03:
I don't understand... Both are working fine now with ble as you can see

[image]

Any idea to debug what happend?

revk, 2024-10-03:
Err no. I mean they do s/w updates automatically, and a release was issued recently, so maybe just that you are on more up to date s/w now perhaps.

closed 2024-10-03 by revk