revk/xESP32-Faikout #515: Need help for connection

Idaho947, 2024-10-04:
Hello,

I need some help to connecte my faikin on my new daikin perfera. I open the split and dont know where I can link the cable.

[image]

Someone can help me ?

revk, 2024-10-04:
The port on the left labelled S21.

Idaho947, 2024-10-04:
Thanks for you very quick reply. I bought a s21 cable but it is to long : 
[image]

[image]

revk, 2024-10-04:
The port on the image you posted definitely has 5 pins.

[image]

Idaho947, 2024-10-04:
Arf need new glasses... And my cable is good too ? Do you have some good link to buy some good cable ?

revk, 2024-10-04:
I know a lot of people just use jumper wire. But I don't have any links, someone else may.

Idaho947, 2024-10-04:
Thanks very much for your reply. Look at the jumper.

Idaho947, 2024-10-23:
Hello those one are nice :
https://fr.aliexpress.com/item/1005006291181398.html?spm=a2g0o.order_list.order_list_main.5.21ef5e5bQxs1az&gatewayAdapt=glo2fra

Idaho947, 2024-10-23:
Found good s21 wire

closed 2024-10-23 by Idaho947