revk/xESP32-Faikout #516: [BUG] FTXC35DV1B can read data but cannot send

aizerin, 2024-10-04:
**Faikin hardware**
My own HW. I cannot easily buy Faikin in my country.
ESP32-WROOM-32D
Faikin S1 build

[image]
[image]
[image]

**Daikin hardware**
FTXC35DV1B
CN_S21

**Describe the bug**
I can see all the data from the uni in web UI, but I cannot control the unit.

I turned on debug I'm getting this message.
```
{"protocol":"S21","mismatch":true,"data":"024700000000004703"}
```
from info cannot-tx
```
{"protocol":"S21","cmd":"D1","payload":"31344635","text":"14F5"}
```

**Additional context**

I tried another board and double-checked the wires. 

What's strange is that when I use GPIO TX:1 and RX:3 as it should be I get system offline. If I swap these for TX:3 and RX:1 then Im able to read the data from the unit.

revk, 2024-10-04:
Tx and Rx are always in the eye of the beholder, so not surprise. Mismatch is an odd error.

You can lock down to not try other protocols like no.x50a, etc, and tell it not to invert tx and rx (which is also tries) if you are sure of polarity.

aizerin, 2024-10-04:
I tried that settings and result it same

revk, 2024-10-04:
And nothing else connected as well? mismatch is a very odd message.

aizerin, 2024-10-04:
What do you mean by nothing else? I have only AC unit, remote controller and faikin. I have a JST plug in S21 so the wires should be secured.

I also tried to use uart2 and the situation was the same.

aizerin, 2024-10-04:
What should I send? I turned on dump and debug.

there are a lot of records in tx and rx.

TX
```
{"protocol":"S21","dump":"0246418703","FA":""}
{"protocol":"S21","dump":"0246317703","F1":""}
{"protocol":"S21","dump":"0246367C03","F6":""}
...
```

RX
```
{"protocol":"S21","dump":"024733303030303A03","G3":"0000"}
{"protocol":"S21","dump":"024700000000004703","G":"\u0000\u0000\u0000\u0000"}
{"protocol":"S21","dump":"024732343A8080E703","G2":"4:\u0080\u0080"}
...
```

revk, 2024-10-04:
Rx looks messy, you sure you have the levels right? Daikin wants 5V typically.

aizerin, 2024-10-04:
It should be fine. I have a level shifter - I tried two different boards to be sure that there was no failure.

aizerin, 2024-10-04:
I have tested a few devices

esp-wroom-e32 - system offline
esp8266 - system offline
esp-wroom-e32d - only read

I will order level shifter from a different brand and esp32 s3.

Sonic-Amiga, 2024-10-07:
Hello!
So, you are not getting data from the AC to controller, correct ?
Remember that these transistor-based converters require +5V pull-up; and you may not have it from the A/C
Here's a tested solution: https://github.com/Sonic-Amiga/ESP8266-Faikin/tree/main/Hardware/S21Adapter . I am actually using this on one of my two units, works like a charm despite utter simplicity. Tx (from Faikin) is inverted, Rx is not.

Yes, that's a 8266 repo and indeed i am using this contraption with a ESP32 board.

aizerin, 2024-10-09:
The problem was with the level shifters I had. 

I bought a new SparkFun BOB-12009 and it works fine so far.

closed 2024-10-09 by revk