revk/xESP32-Faikout #521: [FEATURE] MQTT - Add hostname to state topic

tobelix87, 2024-10-11:
Would it be possible to publish the _hostname_ in the state topic like it is done for _id_ and _app_ ?

I am currently set my faikin up to work with fhem and it would be nice to pick the hostname automatically when it is provided via mqtt topic. Otherwise I would need to manually add it for every faikin device.

revk, 2024-10-11:
Is *hostname* not in the topic it sends though?

Do you have example of what you see, and what you want, in more detail?

tobelix87, 2024-10-11:
I receive under Faikin/EG_WZ_Faikin/state the following (I have prefixapp and prefixhost enabled and "EG_WZ_Faikin" is my hostname). Don't be confused by the ***s, I was just anonymizing some values:

`{
  "ts": "2024-10-11T12:23:18Z",
  "ts": "2024-10-11T15:47:01Z",
  "id": "DC5475****",
  "up": true,
  "uptime": 574092,
  "mqtt-up": 15217,
  "uptime": 586315,
  "mqtt-up": 4246,
  "app": "Faikin",
  "version": "ebf3c2b",
  "build-suffix": "-S3-MINI-N4-R2",
  "build": "2024-09-26T14:35:29",
  "flash": 4194304,
  "rst": 3,
  "mem": 120884,
  "mem": 120892,
  "spi": 2090336,
  "ssid": "xyz",
  "bssid": "7C****F1",
  "rssi": -49,
  "chan": 6,
  "rssi": -46,
  "chan": 1,
  "ipv4": "192.168.173.210"
}`

here I would like to see the hostname as well.

revk, 2024-10-11:
OK but surely `Faikin/EG_WZ_Faikin/state` includes the *hostname* `EG_WZ_Faikin` so why would it need to be in the payload as well?

tobelix87, 2024-10-30:
you are right.
I have thought it through again and I will always need to know the hostname before I can get the payload via MQTT, so it makes not much sense to add it.

closed 2024-10-30 by tobelix87