revk/xESP32-Faikout #526: [FEATURE] Enable hotspot when device can not connect to WiFi

Y0Sam, 2024-10-16:
My Faikin is working fine to begin with, so thanks for that!

I was connected to one of my Faikins for updating the firmware and changing the MQTT connection settings.
Accidentally I copied the new password for the MQTT connection in the WiFi password field which resulted in a lost connection.
To solve this I needed to re-flash the unit.

To me it would make sense to activate the access point of the Faikin, when it can not connect to the WiFi network of the setup.
This would enable the possibility to fix the connection issue without having to flash the device.

Regards

revk, 2024-10-16:
That is exactly what it does though. If it cannot connect to WiFi it starts an access point.

To be clear this is failing to connect to WiFi, not if it can connect on WiFi but fails to connect on MQTT. In that case you can simply access its web interface on whatever IP it has been assigned.

Y0Sam, 2024-10-16:
It is not what I experienced. The WiFi SSID was correct the password not. I restarted the device a couple of times, and I did not get the Access Point nor the web interface. So don't know what went wrong there. 
Maybe because I was on a very old firmware version (of about a year ago).
I was able to flash the device so no problem there, I will post a more elaborate programming manual in the discussion section.

revk, 2024-10-16:
The SSID being correct and password wrong should still mean AP mode. But it may constantly be trying if it sees the SSID, so maybe that is an edge case I have not tested.

For now I suggest turn off your wifi (or change SSID) so it goes in to AP mode. Then you can get it at up right.

Y0Sam, 2024-10-16:
Thanks for the suggestion, as stated I flashed the latest firmware, which I was intending to do anyway, as the OTA update did not work (when the device was still connected to the wifi, also on my second device it could not connect to the OTA server. The OTA update has however worked at least once before on both devices.
For now my problems are fixed.
Good luck with testing my edge case.

closed 2024-10-16 by revk