revk/xESP32-Faikout #527: [BUG] ESP32C3 SuperMini build error

yqkee0411, 2024-10-16:
Hi, I have managed to get it working on my esp8266 but when trying to flash on my esp32c3 Supermini it give me error, if changes to esp32s3,the build will succeed.
Is the esp32c3 Mini supported?

**Faikin hardware**
ESP32C3 SuperMini

Environment running: WSL ubuntu

**Describe the bug**
When changing the idf.py set-target to esp32c3, idf.py build will throw error below 

[1040/1051] Building C object esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/revk.c.obj
FAILED: esp-idf/ESP32-RevK/CMakeFiles/__idf_ESP32-RevK.dir/revk.c.obj
/home/yqkee0411/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DBUILD_ESP32_USING_CMAKE -DESP_PLATFORM -DIDF_VER=\"v5.2.3\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/yqkee0411/esp/ESP32-Faikin/ESP/build/config -I/home/yqkee0411/esp/ESP32-Faikin/ESP/components/ESP32-RevK/include -I/home/yqkee0411/esp/esp-idf/components/newlib/platform_include -I/home/yqkee0411/esp/esp-idf/components/freertos/config/include -I/home/yqkee0411/esp/esp-idf/components/freertos/config/include/freertos -I/home/yqkee0411/esp/esp-idf/components/freertos/config/riscv/include -I/home/yqkee0411/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/yqkee0411/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/yqkee0411/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/home/yqkee0411/esp/esp-idf/components/freertos/esp_additions/include -I/home/yqkee0411/esp/esp-idf/components/esp_hw_support/include -I/home/yqkee0411/esp/esp-idf/components/esp_hw_support/include/soc -I/home/yqkee0411/esp/esp-idf/components/esp_hw_support/include/soc/esp32c3 -I/home/yqkee0411/esp/esp-idf/components/heap/include -I/home/yqkee0411/esp/esp-idf/components/log/include -I/home/yqkee0411/esp/esp-idf/components/soc/include -I/home/yqkee0411/esp/esp-idf/components/soc/esp32c3 -I/home/yqkee0411/esp/esp-idf/components/soc/esp32c3/include -I/home/yqkee0411/esp/esp-idf/components/hal/platform_port/include -I/home/yqkee0411/esp/esp-idf/components/hal/esp32c3/include -I/home/yqkee0411/esp/esp-idf/components/hal/include -I/home/yqkee0411/esp/esp-idf/components/esp_rom/include -I/home/yqkee0411/esp/esp-idf/components/esp_rom/include/esp32c3 -I/home/yqkee0411/esp/esp-idf/components/esp_rom/esp32c3 -I/home/yqkee0411/esp/esp-idf/components/esp_common/include -I/home/yqkee0411/esp/esp-idf/components/esp_system/include -I/home/yqkee0411/esp/esp-idf/components/esp_system/port/soc -I/home/yqkee0411/esp/esp-idf/components/esp_system/port/include/riscv -I/home/yqkee0411/esp/esp-idf/components/esp_system/port/include/private -I/home/yqkee0411/esp/esp-idf/components/riscv/include -I/home/yqkee0411/esp/esp-idf/components/lwip/include -I/home/yqkee0411/esp/esp-idf/components/lwip/include/apps -I/home/yqkee0411/esp/esp-idf/components/lwip/include/apps/sntp -I/home/yqkee0411/esp/esp-idf/components/lwip/lwip/src/include -I/home/yqkee0411/esp/esp-idf/components/lwip/port/include -I/home/yqkee0411/esp/esp-idf/components/lwip/port/freertos/include -I/home/yqkee0411/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/yqkee0411/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/yqkee0411/esp/esp-idf/components/lwip/port/esp32xx/include/sys -I/home/yqkee0411/esp/esp-idf/components/nvs_flash/include -I/home/yqkee0411/esp/esp-idf/components/spi_flash/include -I/home/yqkee0411/esp/esp-idf/components/esp_partition/include -I/home/yqkee0411/esp/esp-idf/components/app_update/include -I/home/yqkee0411/esp/esp-idf/components/bootloader_support/include -I/home/yqkee0411/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/home/yqkee0411/esp/esp-idf/components/esp_app_format/include -I/home/yqkee0411/esp/esp-idf/components/esp_bootloader_format/include -I/home/yqkee0411/esp/esp-idf/components/esp_http_client/include -I/home/yqkee0411/esp/esp-idf/components/esp_event/include -I/home/yqkee0411/esp/esp-idf/components/esp-tls -I/home/yqkee0411/esp/esp-idf/components/esp-tls/esp-tls-crypto -I/home/yqkee0411/esp/esp-idf/components/mbedtls/port/include -I/home/yqkee0411/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/yqkee0411/esp/esp-idf/components/mbedtls/mbedtls/library -I/home/yqkee0411/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/home/yqkee0411/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/home/yqkee0411/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/home/yqkee0411/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/home/yqkee0411/esp/esp-idf/components/esp_http_server/include -I/home/yqkee0411/esp/esp-idf/components/http_parser -I/home/yqkee0411/esp/esp-idf/components/esp_wifi/include -I/home/yqkee0411/esp/esp-idf/components/esp_wifi/wifi_apps/include -I/home/yqkee0411/esp/esp-idf/components/esp_phy/include -I/home/yqkee0411/esp/esp-idf/components/esp_phy/esp32c3/include -I/home/yqkee0411/esp/esp-idf/components/esp_netif/include -I/home/yqkee0411/esp/esp-idf/components/esp_timer/include -I/home/yqkee0411/esp/esp-idf/components/driver/include -I/home/yqkee0411/esp/esp-idf/components/driver/deprecated -I/home/yqkee0411/esp/esp-idf/components/driver/analog_comparator/include -I/home/yqkee0411/esp/esp-idf/components/driver/dac/include -I/home/yqkee0411/esp/esp-idf/components/driver/gpio/include -I/home/yqkee0411/esp/esp-idf/components/driver/gptimer/include -I/home/yqkee0411/esp/esp-idf/components/driver/i2c/include -I/home/yqkee0411/esp/esp-idf/components/driver/i2s/include -I/home/yqkee0411/esp/esp-idf/components/driver/ledc/include -I/home/yqkee0411/esp/esp-idf/components/driver/mcpwm/include -I/home/yqkee0411/esp/esp-idf/components/driver/parlio/include -I/home/yqkee0411/esp/esp-idf/components/driver/pcnt/include -I/home/yqkee0411/esp/esp-idf/components/driver/rmt/include -I/home/yqkee0411/esp/esp-idf/components/driver/sdio_slave/include -I/home/yqkee0411/esp/esp-idf/components/driver/sdmmc/include -I/home/yqkee0411/esp/esp-idf/components/driver/sigma_delta/include -I/home/yqkee0411/esp/esp-idf/components/driver/spi/include -I/home/yqkee0411/esp/esp-idf/components/driver/temperature_sensor/include -I/home/yqkee0411/esp/esp-idf/components/driver/touch_sensor/include -I/home/yqkee0411/esp/esp-idf/components/driver/twai/include -I/home/yqkee0411/esp/esp-idf/components/driver/uart/include -I/home/yqkee0411/esp/esp-idf/components/driver/usb_serial_jtag/include -I/home/yqkee0411/esp/esp-idf/components/esp_pm/include -I/home/yqkee0411/esp/esp-idf/components/esp_ringbuf/include -I/home/yqkee0411/esp/esp-idf/components/bt/include/esp32c3/include -I/home/yqkee0411/esp/esp-idf/components/bt/common/osi/include -I/home/yqkee0411/esp/esp-idf/components/bt/common/api/include/api -I/home/yqkee0411/esp/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/home/yqkee0411/esp/esp-idf/components/bt/common/btc/profile/esp/include -I/home/yqkee0411/esp/esp-idf/components/bt/common/hci_log/include -I/home/yqkee0411/esp/esp-idf/components/bt/porting/ext/tinycrypt/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/hr/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/htp/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/prox/include -I/home/yqkee0411/esp/esp-idf/components/bt/host/nimble/n [... cut ...]
/home/yqkee0411/esp/ESP32-Faikin/ESP/components/ESP32-RevK/revk.c: In function 'gpio_ok':
/home/yqkee0411/esp/ESP32-Faikin/ESP/components/ESP32-RevK/revk.c:2200:1: error: control reaches end of non-void function [-Werror=return-type]
 2200 | }
      | ^
cc1: some warnings being treated as errors
[1041/1051] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/Faikin.c.obj
ninja: build stopped: subcommand failed.

**To Reproduce**
run "idf,py set-target esp32c3"
then run "idf.py build" or "make"

Thanks in advance

revk, 2024-10-16:
I don't know enough about C3. Does it not have the pull up and down functions?

yqkee0411, 2024-10-16:
from what I have find out is that the esp32c3 doesn't have the support for the rtc gpio

revk, 2024-10-16:
Ah, D'Oh, missed the RTC aspect. I'll have to work out a change to allow a C3 build then.

yqkee0411, 2024-10-16:
Got it, thanks

revk, 2024-10-16:
Update library and` Makefile` updated to allow `make c3`

Do `make pull` and it should pick up, then `make c3` to make a C3 build.

closed 2024-10-16 by revk

yqkee0411, 2024-10-16:
after flashing it, it doesn't seem to work and i think its boot looping

Below are the messages get from connecting it using serial

ELF file SHA256: ff4b49084a587062

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038069a
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x3fc
load:0x403cc710,len:0x8fc
load:0x403ce710,len:0x26bc
entry 0x403cc710
E (509) RevK: Not using LED GPIO 47
E (556) httpd: httpd_server_init: error in listen (112)

assert failed: gpio_reset_pin gpio.c:442 (GPIO_IS_VALID_GPIO(gpio_num))
Core  0 register dump:
MEPC    : 0x403806ae  RA      : 0x4038bbe8  SP      : 0x3fca1e60  GP      : 0x3f                                                                                                                                                                                                                                                                                                             c94a00
TP      : 0x3fc5b534  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33                                                                                                                                                                                                                                                                                                             323130
S0/FP   : 0x00000080  S1      : 0x00000001  A0      : 0x3fca1e9c  A1      : 0x3f                                                                                                                                                                                                                                                                                                             c96c95
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3f                                                                                                                                                                                                                                                                                                             c9d000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3f                                                                                                                                                                                                                                                                                                             ca1fb7
S4      : 0x3fc96c94  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00                                                                                                                                                                                                                                                                                                             000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00                                                                                                                                                                                                                                                                                                             000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62                                                                                                                                                                                                                                                                                                             613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00                                                                                                                                                                                                                                                                                                             000000
MHARTID : 0x00000000

Stack memory:
3fca1e60: 0x00000000 0x3fca1f48 0x3c153158 0x40390b24 0x3fc969b4 0x3c153158 0x3f                                                                                                                                                                                                                                                                                                             c970e4 0x3c13c891
3fca1e80: 0x3fc969c4 0x3fca1e94 0x3fc969c8 0x3c13c898 0x3fc96c94 0x00323434 0x3f                                                                                                                                                                                                                                                                                                             ca39ec 0x65737361
3fca1ea0: 0x66207472 0x656c6961 0x67203a64 0x5f6f6970 0x65736572 0x69705f74 0x70                                                                                                                                                                                                                                                                                                             67206e 0x632e6f69
[... cut ...]                                                                             000000 0x00000000
3fca2240: 0x00000000 0x00000000 0x00000000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5                                                                                                                                                                                                                                                                                                             a5a5a5 0xa5a5a5a5

Thanks

revk, 2024-10-16:
Ah I may need more GPIO checks for C3

yqkee0411, 2024-10-18:
Got it, thanks

DanHKfree97, 2024-12-29:
hello, i know that this is closed. But i flashed my ESP32C3 with bin files provided in release folder and i got same result ... ESP is bootlooping with same error as shown here. So is there solution or not?

revk, 2024-12-30:
```
assert failed: gpio_reset_pin gpio.c:442 (GPIO_IS_VALID_GPIO(gpio_num))
```

That suggests we are not picking up the valid GPIOs. A change was made, are you using the `betarelease` files?

DanHKfree97, 2024-12-30:
Betarelease does the same thing :/

DanHKfree97, 2024-12-30:
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403819c2
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x3fc
load:0x403cc710,len:0x8f4
load:0x403ce710,len:0x26dc
entry 0x403cc710
E (519) RevK: Not using LED GPIO 47
E (560) httpd: httpd_server_init: error in listen (112)

assert failed: gpio_reset_pin gpio.c:438 (GPIO_IS_VALID_GPIO(gpio_num))
Core  0 register dump:
MEPC    : 0x40381a0a  RA      : 0x4038c994  SP      : 0x3fca37b0  GP      : 0x3fc95c00
TP      : 0x3fca3a50  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000080  S1      : 0x00000001  A0      : 0x3fca37ec  A1      : 0x3fc983b5
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc9f000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fca3907
S4      : 0x3fc983b4  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fca37b0: 0x00000000 0x3fca3898 0x3c153fb8 0x40391b4c 0x3fc98710 0x3c153fb8 0x3fc98c9c 0x3c13df61
3fca37d0: 0x3fc98720 0x3fca37e4 0x3fc98724 0x3c13df1c 0x3fc983b4 0x00383334 0x3fca52c4 0x65737361
3fca37f0: 0x66207472 0x656c6961 0x67203a64 0x5f6f6970 0x65736572 0x69705f74 0x7067206e 0x632e6f69
3fca3810: 0x3833343a 0x50472820 0x495f4f49 0x41565f53 0x5f44494c 0x4f495047 0x69706728 0x756e5f6f
3fca3830: 0x0029296d 0x00000000 0x00000001 0x00000001 0x00000001 0x00000004 0x00000000 0x600c2000
3fca3850: 0x00000001 0x0000000a 0x00000000 0x3fca38f8 0x00000000 0x00000000 0x00000000 0x00000000
3fca3870: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000004 0x000000a8 0x000000c0
3fca3890: 0x00000000 0x00000000 0x3fca81bc 0x4038cdce 0x00000000 0x00000000 0x3fca81bc 0x4211b9a2
3fca38b0: 0x00000005 0x00000000 0x00000000 0x00000000 0x3fc9ee0c 0x00000000 0x00000000 0x00000000
3fca38d0: 0x00000000 0x00000000 0x3fc9ef24 0x4205c07a 0x00000000 0x3fcba54c 0x3fca07c0 0x4038fb64
3fca38f0: 0x00000000 0x3fce0000 0x3fcba54c 0x40381ea4 0x00000000 0x3fce0000 0x3fcba54c 0x420160f4
3fca3910: 0x00000000 0x00000003 0x3fcba54c 0x420cadfc 0x00000000 0x00000000 0x00000000 0x420ccb20
3fca3930: 0x00000000 0x00000000 0x3fc9ef24 0x420167ac 0x00000804 0x0000000c 0x00000000 0x40381b64
3fca3950: 0x3fca41b0 0x00000010 0x00000010 0x40381bc0 0x00000005 0x00001800 0x7fffffff 0x00000804
3fca3970: 0x80000050 0x00130007 0x00050008 0x00050000 0x00000005 0x00000000 0x00000000 0x00000000
3fca3990: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca39b0: 0x00000000 0x00000000 0x00000001 0x4205ee40 0x3fca4314 0x3fca4314 0x3fca3a00 0x4205e786
3fca39d0: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca39f0: 0x00000000 0x00000000 0x00000000 0x4211a5f6 0x00000000 0x00001388 0x00000001 0x00000001
3fca3a10: 0x00000000 0x00000000 0x00000000 0x4038cb08 0x00000000 0x00000000 0x00000000 0x00000000
3fca3a30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca3a50: 0xa5a5a5a5 0x00000150 0x3fca3810 0x0000001e 0x3fc99180 0x3fcb90bc 0x3fca3a58 0x3fc99178
3fca3a70: 0x00000018 0x3fca5598 0x3fca5598 0x3fca3a58 0x00000000 0x00000001 0x3fca2a54 0x6e69616d
3fca3a90: 0x00000000 0x00000000 0x00000000 0x3fca3a50 0x00000001 0x00000000 0x3fca8e48 0x42007992
3fca3ab0: 0x00000070 0x3fca0f68 0x3fca0fd0 0x3fca1038 0x00000000 0x00000000 0x00000001 0x00000000
3fca3ad0: 0x00000000 0x00000000 0x42006ea6 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3af0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3b10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3b30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3b50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3b70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3b90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000600 0xa5a5a5a5

ELF file SHA256: 3df8e0e9ee2d2b7a

revk, 2024-12-30:
I'll have another look.

revk, 2024-12-30:
OK the library has checks for `GPIO_IS_VALID_GPIO` but Faikin code itself does not, so default tx and rx may be invalid.

So adding that to Faikin code and making a beta.