revk/xESP32-Faikout #529: FBQ71C7VEB  : Ducted unit, No S21 or X50A on the diagram

yiannosi, 2024-10-18:
Hi,

I have a  FBQ71C7VEB (ducted unit) where the X50A is not shown on the diagram but spotted on the PCB!

Just as Issue https://github.com/revk/ESP32-Faikin/issues/305#issue-2250068272

Doesn't seem to work !
Anyone found a solution with any of the other ports?
[image]
[image]

Anyone found a solution?

MUN0X, 2024-10-18:
Did you combine it X50A with power from X35A as explained in the [wiki](https://github.com/revk/ESP32-Faikin/wiki/Wiring) ?

Its a P1P2 unit so perhaps P1P2MQTT could be an option if you can't get faikin to work?

yiannosi, 2024-10-19:
> Did you combine it X50A with power from X35A as explained in the [wiki](https://github.com/revk/ESP32-Faikin/wiki/Wiring) ?
> 
> Its a P1P2 unit so perhaps P1P2MQTT could be an option if you can't get faikin to work?

Yes I connected the X35A as well. Faikin powers up normally, it just stays offline.
The problem is with the Serial connection at X50A.
Not exactly sure what connector it is. It looks like 1.25 pitch JST 8-pin connector but am not sure. I had a 3pin connector instead so since only Rx,TX pins are required, I used that one, see below:

[image]

alexcf, 2024-10-19:
I have a similar unit. X50 that’s currently implemented doesn’t work just yet. Need to learn how to reverse engineer it so we can get it going. Do you have Madoka too?

IMG_20241018_185226.jpg (view on web)

revk, 2024-10-19:
X50 should work. All my ducted units are X50. I coded that before S21 even.

yiannosi, 2024-10-19:
> X50 should work. All my ducted units are X50. I coded that before S21 even.

In Issue https://github.com/revk/ESP32-Faikin/issues/305#issue-2250068272 someone mentioned that they also have n X50 port on he PCB that it doesnt show on the wiring diagram exactly just like my case. They then said that "It seems like the X50A isn't wired in. ".
I wonder if this is the case I am facing or it is some sort of a bad wiring issue .

yiannosi, 2024-10-19:
Madoka? No, mine are older conrrollers BRC1D52.

Is it a case were the X50 is there but disabled? 
If this is the case, then there is no solution. I am still not sure if the rx, tx pins contact properly  on the X50 as i dont have the proper 8 pin plug. See pic above.

alexcf, 2024-10-19:
I’m getting some kind of output from mine.

yiannosi, 2024-10-20:
Just  tested mine and i get 5v in both rx and tx coming from the X50A which  doesnt seem right.
I ve read  somewhere  that there are might be different  pinouts for the x50a.

revk, 2024-10-20:
5V is right as driving one and pulling up the other.

Sonic-Amiga, 2024-10-21:
Read the thread and checked out p1p2 project. No, that's a differential bus with complex signaling; it's not CN_WIRED. BTW, Faikin H/W cannot do P1P2 in principle

yiannosi, 2024-10-22:
> Read the thread and checked out p1p2 project. No, that's a differential bus with complex signaling; it's not CN_WIRED. BTW, Faikin H/W cannot do P1P2 in principle

Correct. I realise  that now. P1P2 is not an option

dzungpv, 2024-10-25:
On this model X50A exist but they disable by the firmware, so you solution is try P1P2 communication or IR if they have IR remote. P1P2 require difference hardware, it is quite complicate. You can check P1P2MQTT project. I have boards for P1P2 running ESP32 but is is a commercial project.

yiannosi, 2024-10-25:
> On this model X50A exist but they disable by the firmware, so you solution is try P1P2 communication or IR if they have IR remote. P1P2 require difference hardware, it is quite complicate. You can check P1P2MQTT project. I have boards for P1P2 running ESP32 but is is a commercial project.

Thanks very much for the info.

Where did you get this model-specific information  from?

Sonic-Amiga, 2024-11-05:
Hello everyone!
Just FYI, here https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/11#issuecomment-2454487473 is pinout of X50A cable, which came with BRP069C41. Yes, small X50a connector is tricky.

yiannosi, 2024-11-06:
> Hello everyone! Just FYI, here [Sonic-Amiga#11 (comment)](https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/11#issuecomment-2454487473) is pinout of X50A cable, which came with BRP069C41. Yes, small X50a connector is tricky.

Thanks for the pinout.

[https://github.com/revk/ESP32-Faikin/issues/529#issuecomment-2437380173](url) said that above model the X50A port although present, is disabled.

yiannosi, 2024-11-06:
> > > Hello everyone! Just FYI, here [Sonic-Amiga#11 (comment)](https://github.com/Sonic-Amiga/ESP8266-Faikin/issues/11#issuecomment-2454487473) is pinout of X50A cable, which came with BRP069C41. Yes, small X50a connector is tricky.
> > 
> > 
> > Thanks for the pinout. **dzungpv** above suggested that above model the X50A port although present, is disabled. Can anyone confirm?
> 
> I confirm that, I have try 1 AC with X50a too, no data at 9600 baud as we decode in this project, not same model but same duct type.
> 
> But seem you are wiring it wrong: see this pic in wiki: https://github.com/revk/ESP32-Faikin/wiki/Wiring the bottom left image . See this pic to convert from S21 to X50a and X35a: https://user-images.githubusercontent.com/44964969/62939548-c1f4d600-bdfb-11e9-95c8-4628c0121dcd.jpg

Yes. The dfference in my connection is that I dont use Pin 1 (5V) connection since my Faikin generates 5V onboard. So I just use pin 3 &4 from the X50A connector only. (The red wire shown in my pic, is not connected. Its just that I happned to have a 3pin jst connector)

dzungpv, 2024-11-06:
Also try to swap the tx and rx, make sure you board working too.

yiannosi, 2024-11-10:
Update:
I bought a per-crimped  8-pin ZH1.5 mm plug and tried again with proper wiring.
Faikin connects to WiFi but unit stays offline as before.
I think it is now safe to deduce that this model, although it has an X50A connector, it does not work.