revk/xESP32-Faikout #533: [FEATURE] add govee 5075 to ble sensor

Idaho947, 2024-10-23:
**Daikin hardware**
Daikin perfera 2024 on s21 port

**Is your feature request related to a problem? Please describe.**
My blea govee 5075 sensor is not show in faikin web interface

**Describe the solution you'd like**
Add the compatibility with rhis sensor :

https://eu.govee.com/products/govee-bluetooth-hygrometer-thermometer-h5075?srsltid=AfmBOoprjkwyaOBJr5EUgHgT7ko3oX_-L4ZqOhkSx5EK--F6fGAMakqO

revk, 2024-10-24:
I'll have to look in to it, there is another model which does work. So may not be too hard, but I won't know until I get one.

revk, 2025-12-04:
Sorry, snowed under.

Evolutionary, 2025-12-12:
I have had Govee H5075 connected via BLE with each of my Faikout units since installing/updating them in late October.