revk/xESP32-Faikout #537: [BUG] System is offline.

TechnicGuy1, 2024-10-25:
**Faikin hardware**
from Amazon

**Daikin hardware**
S21 ATXC50

**Describe the bug**
Faikin Webgui say system is offline

Hello,

after i thought some wifi problems, restart Access Point, and again no webgui from Faikin. Set power off of the Daikin ATXC50. After some shutdown's, webgui is here. But "System is offline.", all cable are unchanged.
Any hints to get it "connected/online state" ?

Thanks, Best Regards Michael

revk, 2024-10-25:
Not sure what to suggest - turn on debug and check MQTT maybe?

TechnicGuy1, 2024-10-31:
After several reboots, faikin show normal, but no interaction on demand or temperature changes. Shutdown, and reconnectet all cable, and its running. Thanks ;)

closed 2024-10-31 by TechnicGuy1