revk/xESP32-Faikout #542: [BUG] Some sensors are missing since <ebf3c2b> in HA (missing discovery topics)

con0aks, 2024-11-05:
**Faikin hardware**
Faikin-S3-MINI-N4-R2

**Daikin hardware**
FTXM35R (S21)

**Describe the bug**
I have three faikin boards running. I recently updated one of the boards.
Since the version <ebf3c2b 2024-09-26T14:35:29> some sensors are missing in HA (liquid temp, comp freq, fan freq).

**To Reproduce**
Steps to reproduce the behavior:
-

**Expected behavior**
On the other boards I got five sensors (liquid temp, comp freq, fan freq, outside temp, lifetime energy) in HA.

**Additional context**
The sensor values are present in the mqtt-topics, but the configs for discovery are missing.

TheBestMoshe, 2024-11-05:
[I had an issue similar to this](https://github.com/revk/ESP32-Faikin/discussions/519).

My solution was to turn off the home assistant option on the Faikin, and then reenable it. That caused it to be shown again.

con0aks, 2024-11-05:
Thanks for the advise.

Sadly I tried it already... didn't helped.

I also disabled mqtt and reenabled it after deleting all topics and restarting home assistant.
But also, no success.

revk, 2024-11-06:
Can you confirm they do show on the web UI?

Have you tried restarting MQTT.

Try deleting the device from HA as well, and then restarting Faikin?

Finally in HA under MQTT you can get it to show what it received as the auto config, can you check that and post a copy.

con0aks, 2024-11-07:
I can still see all values / sensors on the web interface.
Restarting MQTT and deleting the device in HA does not help.

But: When I disable and re-enable HA in the config of the faikin board I can shortly see the relevant topics for discovery but the configs of the missing sensors are empty.

I'm not exactly sure what you mean with your last point. Could you please explain it a little bit further.

Thanks in advice.

revk, 2024-11-08:
HA let's you see the config that is announced.

Check that.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

con0aks, 2024-11-08:
Ah, got it :)

Here are my results:
[image]
[image]
[image]
[image]
[image]

revk, 2024-11-08:
OK that shows liquid.
You say the web shows the comp freq, and fan freq though?

con0aks, 2024-11-08:
you are right:
Now the liquid sensor is present but the outside is gone...

Not exactly:
When the device is turned off: I can only see the temps.
When the device is turned on: I can additionally see the fan speed.

I cannot set the comp frequency on the web ui.

[image]
[image]

revk, 2024-11-08:
Err, "cannot set the comp frequency", correct, there is no setting for that. There is a demand setting which has some impact.

This is all very odd.

revk, 2024-11-08:
> Where did you saw that message?

Who, what message?

con0aks, 2024-11-08:
"cannot set the comp frequency"

con0aks, 2024-11-08:
Sry, got a little confused :D

revk, 2024-11-08:
[image]

revk, 2024-11-08:
Turn on `dump` setting and log the MQTT messages produced.

revk, 2024-11-08:
Ok may be tomorrow when I check this.

con0aks, 2024-11-08:
ok, thanks in advance.

revk, 2024-11-09:
OK you are getting compressor. So I am wondering if it is a timing thing, HA config sent too soon before it has seen compressor, etc.

Ah, I worked it out, we changed to poll some things more slowly, and we do the HA config after the first poll, so some things not yet polled, D'Oh.

I'll make a beta release for you to test. Ready in a few minutes.

con0aks, 2024-11-09:
Hey, I just upgraded the board to the new beta and now all five sensors are back in HA.

Thanks a lot for your rapid help!

closed 2024-11-09 by con0aks