revk/xESP32-Faikout #543: Compatibility Question

jax8158, 2024-11-07:
I have a daikin air conditioner, model number FDXS12DVJU. I was wondering if it was compatible with the project? I am pretty new to this project and I don't quite understand so if someone could help me clarify before I spend a bunch of money to send this to America. Thanks for all help in advance! (Below I have included an image of the S21 Plug.) [image]

reedy, 2024-11-12:
https://github.com/revk/ESP32-Faikin/wiki/List-of-confirmed-working-air-con-units does list `FDXS25EAVMB` and `FDXS50CVMB`, so it seems the same base `FDXS` might suggest you're in luck.

Is the S21 being currently used by the existing daikin wifi module?

jax8158, 2024-11-12:
no, just the IR module that came with it, the reason why I want to use this is because the daikin wifi module is several hundred dollars.

revk, 2024-11-12:
Can't guarantee it, but any model with S21 port should work with Faikin.

closed 2024-11-13 by jax8158