revk/xESP32-Faikout #549: [BUG] HomeAssistant outside sensor only updates when the unit is on

lochstar, 2024-11-14:
**Faikin hardware**
Official Faikin

**Daikin hardware**
S21 connector to `FTXV60WVMA`

**Describe the bug**
The HomeAssistant `_outside` sensor only appears to update when the AC is running. The temperature always reports the last recorded temperature when the unit was last switched on.

This differs to the HomeAssistant "Daikin" integration which updates the outside temperature periodically.

**To Reproduce**
1. Keep unit powered off.
2. Note the outside temperature never updates.
3. Switch unit on.
4. Outside temperature updates.

**Expected behavior**
Outside temperature sensor should update on a set interval.

revk, 2024-11-14:
Can you enable dump setting and get details.

It may simply be the Daikin does not send updates when not on.

MartB, 2024-11-14:
Works on my FTXM35A5V1B AC so it must be AC dependent.

lochstar, 2024-11-14:
Thansk @MartB 

Issue occurs on both of my units `FTXV60WVMA` and `FTXV35UVMA`.

Strangely they both randomly update the value during the night, at different times.

I'll try to get some data.

lochstar, 2024-11-14:
Alrighty, here's what the polling looks like when idle for `FTXV60WVMA`

```
state/big-ac/status {"protocol":"S21","online":true,"home":22.5,"heat":false,"fanrpm":0,"comp":0,"outside":21.5,"liquid":22.5,"anglev":116,"Wh":1817000,"power":false,"mode":"C","temp":23.0,"demand":100,"fan":"A","swingh":true,"swingv":true,"econo":false,"powerful":false,"comfort":false,"streamer":false,"sensor":false,"quiet":false,"autor":0.0,"autot":0.0,"auto0":"00:00","auto1":"00:00","autop":false}
info/big-ac/tx {"protocol":"S21","dump":"0246317703","F1":""}
state/big-ac {"ts":"2024-11-14T21:52:53Z","id":"34B7DA58E878","up":true,"uptime":4136,"mqtt-up":4133,"app":"Faikin","version":"0c206eb","build-suffix":"-S3-MINI-N4-R2","build":"2024-11-13T20:43:07","flash":4194304,"rst":3,"mem":121080,"spi":2090236,"ssid":"**","bssid":"E663DAAA9157","rssi":-38,"chan":6,"ipv4":"***","online":true,"power":false,"target":23.00,"temp":22.50,"outside":21.50,"liquid":22.50,"demand":100,"Wh":1817000,"fanfreq":0.0,"comp":0,"mode":"off","fan":"auto","streamer":false,"quiet":false,"econo":false,"comfort":false,"powerful":false,"sensor":false,"swing":"H+V","preset":"home"}

info/big-ac/tx {"protocol":"S21","dump":"0246317703","F1":""}
info/big-ac/rx {"protocol":"S21","dump":"02473130334A416603","G1":"03JA"}
info/big-ac/tx {"protocol":"S21","dump":"0246357B03","F5":""}
info/big-ac/rx {"protocol":"S21","dump":"024735373F3080A203","G5":"7?0\u0080"}
info/big-ac/tx {"protocol":"S21","dump":"0246367C03","F6":""}
info/big-ac/rx {"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"0246377D03","F7":""}
info/big-ac/rx {"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"02464D9303","FM":""}
info/big-ac/rx {"protocol":"S21","dump":"02474D414636348503","GM":"AF64"}
info/big-ac/tx {"protocol":"S21","dump":"0252479903","RG":""}
info/big-ac/rx {"protocol":"S21","dump":"02534741DB03","SG":"A"}
info/big-ac/tx {"protocol":"S21","dump":"0246317703","F1":""}
info/big-ac/rx {"protocol":"S21","dump":"02473130334A416603","G1":"03JA"}
info/big-ac/tx {"protocol":"S21","dump":"0246357B03","F5":""}
info/big-ac/rx {"protocol":"S21","dump":"024735373F3080A203","G5":"7?0\u0080"}
info/big-ac/tx {"protocol":"S21","dump":"0246367C03","F6":""}
info/big-ac/rx {"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"0246377D03","F7":""}
info/big-ac/rx {"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"02464D9303","FM":""}
info/big-ac/rx {"protocol":"S21","dump":"02474D414636348503","GM":"AF64"}
info/big-ac/tx {"protocol":"S21","dump":"0252489A03","RH":""}
info/big-ac/rx {"protocol":"S21","dump":"0253483532322B5F03","SH":"522+"}
info/big-ac/tx {"protocol":"S21","dump":"0246317703","F1":""}
info/big-ac/rx {"protocol":"S21","dump":"02473130334A416603","G1":"03JA"}
info/big-ac/tx {"protocol":"S21","dump":"0246357B03","F5":""}
info/big-ac/rx {"protocol":"S21","dump":"024735373F3080A203","G5":"7?0\u0080"}
info/big-ac/tx {"protocol":"S21","dump":"0246367C03","F6":""}
info/big-ac/rx {"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"0246377D03","F7":""}
info/big-ac/rx {"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"02464D9303","FM":""}
info/big-ac/rx {"protocol":"S21","dump":"02474D414636348503","GM":"AF64"}
info/big-ac/tx {"protocol":"S21","dump":"0252499B03","RI":""}
info/big-ac/rx {"protocol":"S21","dump":"0253493033322B5C03","SI":"032+"}
info/big-ac/tx {"protocol":"S21","dump":"0246317703","F1":""}
info/big-ac/rx {"protocol":"S21","dump":"02473130334A416603","G1":"03JA"}
info/big-ac/tx {"protocol":"S21","dump":"0246357B03","F5":""}
info/big-ac/rx {"protocol":"S21","dump":"024735373F3080A203","G5":"7?0\u0080"}
info/big-ac/tx {"protocol":"S21","dump":"0246367C03","F6":""}
info/big-ac/rx {"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"0246377D03","F7":""}
info/big-ac/rx {"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"02464D9303","FM":""}
info/big-ac/rx {"protocol":"S21","dump":"02474D414636348503","GM":"AF64"}
info/big-ac/tx {"protocol":"S21","dump":"025261B303","Ra":""}
info/big-ac/rx {"protocol":"S21","dump":"0253613531322B7703","Sa":"512+"}
info/big-ac/tx {"protocol":"S21","dump":"0246317703","F1":""}
info/big-ac/rx {"protocol":"S21","dump":"02473130334A416603","G1":"03JA"}
info/big-ac/tx {"protocol":"S21","dump":"0246357B03","F5":""}
info/big-ac/rx {"protocol":"S21","dump":"024735373F3080A203","G5":"7?0\u0080"}
info/big-ac/tx {"protocol":"S21","dump":"0246367C03","F6":""}
info/big-ac/rx {"protocol":"S21","dump":"024736303030303D03","G6":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"0246377D03","F7":""}
info/big-ac/rx {"protocol":"S21","dump":"024737303030303E03","G7":"0000"}
info/big-ac/tx {"protocol":"S21","dump":"02464D9303","FM":""}
info/big-ac/rx {"protocol":"S21","dump":"02474D414636348503","GM":"AF64"}
info/big-ac/tx {"protocol":"S21","dump":"02524C9E03","RL":""}
info/big-ac/rx {"protocol":"S21","dump":"02534C3030302F03","SL":"000"}
info/big-ac/tx {"protocol":"S21","dump":"0246317703","F1":""}
info/big-ac/rx {"protocol":"S21","dump":"02473130334A416603","G1":"03JA"}
info/big-ac/tx {"protocol":"S21","dump":"0246357B03","F5":""}
[... 25 lines cut ...]
```
command/big-ac/send "F2"
[... cut ...]
info/big-ac/rx {"protocol":"S21","cmd":"FY","payload":"3130","text":"10","nak":true}

command/big-ac/send "VS000M"
info/big-ac/rx {"protocol":"S21","dump":"02565331303537303035314D30303030307903","VS":"10570051M00000"}
error/big-ac/comms {"protocol":"S21","mismatch":true,"data":"02565331303537303035314D30303030307903"}
```

Seems to be protocol v2. I can see the outside temp being returned but it doesn't update when idle. I need to do further testing when the unit is operating. I wonder if there's a way to force query it in order to update the value.

revk, 2024-11-15:
Might be better debug not dump, sorry, might be clearer. I'll have to look up which message we are looking for.

But I think it is down to what the Daikin does.

MuzzaMcD, 2024-11-15:
Does this help ... I was looking into this outside temp issue a little while ago
https://github.com/revk/ESP32-Faikin/discussions/220
Regards

lochstar, 2024-11-15:
@MuzzaMcD Thanks for this! I have the exact same model controllers. I just tried sending the `DH1000` command and it updated the outside temperature reading!

Amazing work. Was there ever any discussion to integrate this into Faikin for supported units? The Wiki curently has nothing listed for the [DH commands](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol#dh-command).

I can easily add a poll in HomeAssistant for this anyway.

MuzzaMcD, 2024-11-16:
> add a poll in HomeAssistant

Yep that is what I did. I send DH1000 every 30 mins to just one of the units on the West side of my house IF  MODE=OFF for more than 5 mins. 

```
alias: Test DH0000
description: "When Unit OFF, poke every 30 mins for outside temp update "
mode: single
triggers:
  - minutes: /30
    trigger: time_pattern
conditions:
  - condition: state
    entity_id: switch.ac_study_power
    state: "off"
    for:
      hours: 0
      minutes: 5
      seconds: 0
actions:
  - data:
      qos: 0
      topic: command/AC_Study/send
      payload: DH1000
    action: mqtt.publish
```
At the time I did not do more than 'discover' the DH1000 command (which was a challenge in itself) and did not do any real work looking into what the total reason for its existence.  
All I know is that with the original Daikin controller fitted, the DH1000 command was issued hourly. I don't actually know if it does it all the time or just when the unit is MODE=OFF or indeed what else it does other than result in the outside temp sensor updating mysteriously.    Anyway glad it helped :)

revk, 2024-11-16:
I am sure we can add DH

MuzzaMcD, 2024-11-16:
> I am sure we can add DH

First, Im happy doing this in HA automation at the moment 'cause **my** use case is solved. . . . however

I can also see it might be useful for others for it to be included in Fakin code, because after all if we are trying to mimic the original Daikin controller it appears to issue this signal every hour. 
Maybe (my suggestion only, others could comment) ... that it is optional, and that maybe we allow a timer selection 30/60 min?
Note that when this DH1000 command is sent, the unit does use a little extra power for a few minutes before it goes back to sleep? See below  my unit waking up every 30 mins yesterday.

[image]

Meanwhile, here in the 'south' I have just switched my AC from Heat to Cool .... outside temp reached 38C this afternoon.
Regards

Sonic-Amiga, 2024-11-19:
Wow, interesting observation about DH command. Worth adding to wiki i guess.

closed 2024-11-19 by lochstar