revk/xESP32-Faikout #552: [BUG] "System is offline" after connecting GPIO leads directly to S21 Rx/Tx

Jolt151, 2024-11-17:
**Faikin hardware**
Using this ESP32 dev board, flashed with `Faikin-S1` from `release/` https://www.amazon.com/dp/B0DG8JFY3C 

There's a pinout diagram on the amazon page.

**Daikin hardware**
Daikin `ctxs07lvju`, via the S21 port

**Describe the bug**
Pretty straightforward - I flashed the dev board, setup Faikin through the captive portal and connected it to MQTT, updated the Rx/Tx pins on the settings page to 18/19 (Tx=19, Rx=18), connected those via jumper wire from the board to pins 2 and 3 on the S21 port, and powering the board via the built in USB port. There are no other jumper wires connected

**Expected behavior**
Controls should work?

**Additional context**
Most important context is that I'm new to microcontroller/ESP32 dev, so I don't know what I'm doing. Based on whatever I read, my understanding was that Rx/Tx are the only pins required for communication, so if I power it via the USB port then I can figure out how to connect those pins later, after confirming it works. I'm able to read logs from MQTT, here's a sample:

```
Topic: error/faikin-1/commsQoS: 0
{
  "protocol": "S21¬Tx¬Rx",
  "cmd": "F1",
  "noack": true,
  "value": "46"
}

Topic: error/faikin-1/commsQoS: 0
{
  "protocol": "S21¬Tx¬Rx",
  "cmd": "F1",
  "noack": true,
  "value": "46"
}

Topic: error/faikin-1/commsQoS: 0
{
  "protocol": "loopback¬Tx¬Rx",
  "cmd": "F1",
  "noack": true,
  "value": "FF"
}
2024-11-17 14:43:19:500

Topic: error/faikin-1/commsQoS: 0
{
  "protocol": "loopback¬Tx¬Rx",
  "cmd": "F1",
  "noack": true,
  "value": "FF"
}
```

Any ideas? Is it ok to be connecting these leads directly to the port? I appreciate all help, and I can get any logs necessary. Thanks in advance!!

Jolt151, 2024-11-17:
Some images:

[IMG20241117144657.jpg](https://github.com/user-attachments/assets/dcbacce1-eca6-4877-8758-6330abecd3e6)

[IMG20241117144650.jpg](https://github.com/user-attachments/assets/bb86b2e9-9a4d-4fba-b1e2-f65d6642bf77)

revk, 2024-11-17:
And GND?!?!?!?!?!?

Jolt151, 2024-11-17:
Thanks for the response! Tried adding ground and it still wasn't working, then my genius was wondering if i mixed up the order of the RX/TX pins, and connected it to 3/4 instead, which I think burnt out my board. I'll be getting a new one and trying again.

Sonic-Amiga, 2024-11-19:
Hello! 3.3V level from the ESP appears not enough for S21 port, A/C does not hear Faikin. You need level conversion at least on Tx (Faikin side, to conditioner) line. Been there, done that, and came up with this very simple schematic: https://github.com/Sonic-Amiga/ESP8266-Faikin/tree/main/Hardware/S21Adapter.

You can verify your board by connecting Rx and Tx together, Faikin should enter "loopback test mode" and report it in the UI.
Even if you botched your pins, ESP32 is very flexible and has a matrix switch, so you can program any pins you want. Manuals claim ESPs are 5V-tolerant.

So, unless you plugged it into +14V power, you should be okay. BTW, note that you most likely need a step-down converter, because apparently +5V from the AC on S21 port (if exists) is too weak to provide power for the board; this has also been tested.

Jolt151, 2024-11-21:
Thanks @Sonic-Amiga! Got new boards and I think this matches what I'm seeing: S21 Rx returns data in the `value` field, but S21 Tx returns `00`

```
2024-11-20 22:41:37:143

Topic: error/faikin-1/commsQoS: 0
{
  "protocol": "S21¬Tx",
  "cmd": "F1",
  "noack": true,
  "value": "00"
}
2024-11-20 22:41:37:146

Topic: error/faikin-1/commsQoS: 0
{
  "protocol": "S21¬Rx",
  "cmd": "F1",
  "noack": true,
  "value": "FE"
}
```

I'll try figuring out how to step up the voltage - still a new field for me so I am struggling to understand your schematic but I will research more. Thanks!

Jolt151, 2024-11-21:
Ordered some 3.3v-5v level shifters and will try when I get them.

Sonic-Amiga, 2024-11-24:
@Jolt151 "S21¬Tx" and "S21¬Rx" are not what you think. ESP32 version also tries to autodetect pin inversion, so it tries for combinations for each protocol:
1. Straight
2. Tx inverted
3. Rx inverted
4. Both inverted.
And that's what you see. Neither of these messages contain any valid reply. "value" is a byte, which arrived instead of ACK. Clearly garbled communication.

>  I am struggling to understand your schematic

It consists of a very primitive inverting level shifter on Tx line and integrated step-down converter chip for power. Nothing more.

Jolt151, 2024-11-25:
Tried out the level shifter and I think I'm progressing, but still getting similar output of values and `noack: true`.

The wiring seems correct to me - 5v connected to HV, 3.3v to LV, GPIO 18/19 to LV1/2, and output to the unit on HV1/2. Ground is connected as well.

To verify my connections, i used a multimeter and found that the level shifter outputs a constant 3.3v and 5v respectively on all the LV and HV 1/2/3/4 pins. Is this normal? I found some posts online and this might be common behavior for level shifters, but I figured I'll ask since this seems unintuitive. 

[IMG20241125144941.jpg](https://github.com/user-attachments/assets/9ab86430-a6b0-4763-bcae-2f6f695833c5)

[IMG20241125145042.jpg](https://github.com/user-attachments/assets/4128d1e1-3386-47f0-8faf-af4bdf0cd6d7)

[IMG20241125145050.jpg](https://github.com/user-attachments/assets/90ab52c0-9f88-4505-8c06-ebfe32bfbe8f)

Jolt151, 2024-12-25:
Finally got this working! After a bunch of failed attempts, I connected 5v and gnd into the level shifter (HV), and (importantly) re-soldered the level shifter and now everything is working. Thanks for the help.

closed 2024-12-25 by Jolt151