revk/xESP32-Faikout #556: [FEATURE] Use S21 commands RzC3 or RzB2 to determine hvac_action

sm-Fifteen, 2024-11-20:
**Daikin hardware**
Two FTXS12LVJU (S21) in a multi-zone arrangement.

**Is your feature request related to a problem? Please describe.**
Currently, Faikin acts as if a unit that is set to heating mode is heating 100% of the time, both in Home Assistant and in the web UI. It does not expose what the unit is actually doing, whether is is active, idle or, in the case of heatpumps, defrosting.

**Describe the solution you'd like**

From having collected and analyzed a fair amount of S21 data (admittedly only on my own units), I'm pretty confident that commands RzC3 and RzB2 both represent the current state of the unit ([as documented on the wiki](https://github.com/revk/ESP32-Faikin/wiki/S21-Protocol#rzb2-command-unit-state)), and that this could be used to determine an [hvac_action](https://developers.home-assistant.io/docs/core/entity/climate/#hvac-action) status for Home Assistant, and possibly the web UI or the LED on the Faikin boards themselves.

Using RzB2, which would be the most straightforward, the mapping could be as follow:

| RzB3 response | hvac_action value            |
| ------------- | ---------------------------- |
| SzB300        | `OFF`                        |
| SzB380        | `IDLE` (unless fan mode)     |
| SzB2C0        | `HEATING`/`COOLING`/`DRYING` |
| SzB2A0        | `DEFROSTING`                 |
| SzB2E0        | `DEFROSTING`                 |

We would need to use the current mode to determine what the unit is doing between `HEATING`/`COOLING`/`DRYING`. Auto (heat-cool) mode would have to be omitted, since it's not currently possible to rely on what the unit is supposed to be doing while in this mode. 

**Describe alternatives you've considered**
Using Faikin Auto mode, but I prefer sticking with the native controls.

**Additional context**
This has been raised before in #548 and #293, among other places.

revk, 2024-11-20:
I have defrost as a separate state for ducted unit from X50. So that would be what I set.

Separately it would make sense to work out what to report to HA for defrosting even when X50.

sm-Fifteen, 2024-11-21:
> I have defrost as a separate state for ducted unit from X50. So that would be what I set.
> 
> Separately it would make sense to work out what to report to HA for defrosting even when X50.

Just `"havc_action":"defrosting"`. It's not listed in [their MQTT doc](https://www.home-assistant.io/integrations/climate.mqtt/), but [the unit tests](https://github.com/home-assistant/core/pull/121448/files) show that it's what HAss is expecting.

revk, 2024-11-22:
OK I have added vac action defrosting, but not yet looked at the Rz messages to set this using S21.

sm-Fifteen, 2024-11-22:
Do you think the LED on the Faikin board should represent the current mode or should it be changed to show the current action, now that there's a way to know when the unit is active, idle and defrosting?

revk, 2024-11-22:
It sort of does, but could it work at a lower level and do defrost as well, maybe...

sm-Fifteen, 2024-11-26:
Ok, so other people in https://github.com/revk/ESP32-Faikin/issues/408#issuecomment-2496246700 with different S21 protocol versions have confirmed that my interpretation of RzB2 is both consistant across versions and seemingly correct. There is at least one extra bit to RzC3 I don't have on my v0 unit, but we haven't figured out that one yet.

Regardless or RzC3, though, I think out understanding of RzB2 is now complete enough to try using it for `hvac_action`. It has everything we need for per-Faikin-board status reporting, whether in single-zone or multi-zone configuration.

TechnicGuy1, 2024-12-09:
Would be nice to see defrost/status and also compressor frequency also in the WebUI :)

revk, 2024-12-09:
I thought compressor showed, but yeh can look to add some things.

sm-Fifteen, 2024-12-09:
Regarding S21 support for detecting when the unit is idle, active or defrosting, is there any further information I should provide? By now it's pretty clear that RzB2 is the command we should be using for this, and I think we understand its return payload pretty well.

revk, 2024-12-10:
OK, I tested `RzB2` here, and I get `SzB200` regardless of state of unit, heating/cooling/off.

So I don't think this works generically.

revk, 2024-12-10:
* Auto `C310` (cooling)
* Off `C300`
* Heat `C340`
* Cool `C310`
* Dry `C310`
* Fan `C310`

sm-Fifteen, 2024-12-10:
> * Auto `C310` (cooling)
> 
>     * Off `C300`
> 
>     * Heat `C340`
> 
>     * Cool `C310`
> 
>     * Dry `C310`
> 
>     * Fan `C310`

Assuming the target is heating and that you switched between each mode in quick succession, that would check out. Based on the data I have, `SzC340` means the unit is actively heating/cooling, and `SzC310` means the compressor is locked for "3 minutes standby" (which won't prevent fan-only from running). Defrost would be `SzC380` if it's starting/ending or `SzC3C0` (40 + 80) if it's currently active.

sm-Fifteen, 2025-01-04:
Since we have bit 0x08 on RzB2 which normally mirrors the unit's power state, couldn't we just compare it to the known power state from F1 to determine if that RzB2 holds any reliable data (and ignore B2 otherwise)? We'll have to pull from it regardless to know if the current mode is heating, cooling, defrost or fan mode, anyway.

Also, one the Faikin board I was using to test this stuff unexpectedly died last week while I was away from home (on `2024-12-26 09:25Z`), so it might be harder for me to test multizone signals for the near future.

sm-Fifteen, 2025-08-21:
> OK, I tested `RzB2` here, and I get `SzB200` regardless of state of unit, heating/cooling/off.
> 
> So I don't think this works generically.

@revk: What are the units you had tested this on, BTW? esphome-daikin-S21 implemented this a few weeks ago, and RzB2 seems to be pretty reliable across unit release years and protocol versions. So far your units would be the only ones I've heard news of that react to RzB2 like that. It would probably be worth documenting.

sm-Fifteen, 2025-08-21:
> Let me know what I need to add, and I will.

@revk: Oh, I was just asking for your S21 indoor unit model numbers, so I can write down that these don't seem to understand RzB2. Though regarding how to deal with that being supported or not, I think one could simply compare the "Online" bit of SzB2 to the power status reported by another command, and to ignore the return value of everything in SzB2 if the two don't match.

Besides your units, I noticed that they could mismatch if the unit is in some form of invalid/error state. If I set my multizone heads to conflicting modes, for example, one of them will go to a "safe" state (power light blinking, louvres closed, though I believe probing for error codes said "00"/no errors) where RzB2 replies with SzB2 00, as if it was turned off, despite other commands saying the unit is on. So yeah, my suggestion would be to probe RzB2, but ignore its response/assume all the other bits are zero if it reports 0 for the online state when other commands indicate it should be on.

Then we could just use those bits to set hvac_action to its correct value, as described in the table I opened the issue with.

sm-Fifteen, 2025-12-09:
The early winter over here has been pretty intense, and having a sensor reading for this would be very useful to coordinate my baseboards when the heatpump is starting to struggle. The esphome-daikin-s21 people added RzB2 to their polling loop [back in July](https://github.com/asund/esphome-daikin-s21/pull/41) and, as far as I can tell, it's been working rather well on their end.

All that's really needed here to ensure compatibility is to cross-reference the "Online" value (`SzB280`/`0b0000_1***`) with "RA" to determine whether the value of RzB2 is trustworthy, and the "active, normal operation" value (`SzB2C0`/`0b0000_1100`) with "RB" to determine what mode the unit is in, and therefore what the havc_action to report should be.

asund, 2026-01-19:
Chiming in from that project: A couple users found it's not always reliable or supported. The code prefers Rg for reporting active and F6 preferred for powerful. Unfortunately I've had to blacklist use on certain models (using M response) because the active bit toggles for them and messes up action reporting. Wish there was a better solution. Looking at the value returned with the bare `Rz` there appears to be a correlation with the third byte ('0' is idle, and at least '4' for actively heating). Though this is my unit that does support RzB2.

It's at least an avenue of future investigation and shows some promise because Rz is listed in the official Daikin controller polling loop.

sm-Fifteen, 2026-01-19:
> Looking at the value returned with the bare `Rz` there appears to be a correlation with the third byte ('0' is idle, and at least '4' for actively heating). Though this is my unit that does support [...]

Bare Rz is unfortunately useless, as it's actually just interpreting the CRC and (IIRC) ETX bytes as part of its expected payload, which usually (not always) ends up aliasing to RzC3 or RzB2 (which is actually how I figured out that they existed). Our wiki merely lists it as supported, as in it doesn't get NAK'd and returns some kind of payload.

That's actually the tricky part with Rz, any 2 bytes payload sent to it, ASCII or not, and even if you didn't think you sent any, is going to be accepted and return *something*. In practice, all unsupported values behave as "floating", where they'll all take the same value, which is unpredictable and varies on its own, seemingly not corresponding to anything. I didn't expect the floating value to actually go high enough to flip the "online" bit, though, which is unfortunate.

asund, 2026-01-20:
I see, I was misreading the protocol documentation and overlooked that it was supported but not part of the polling loop.