revk/xESP32-Faikout #559: [BUG] GPIO connection to the board with usb-serial adapter

rici44, 2024-11-21:
**Faikin hardware**
[Amazon purchased](https://www.amazon.co.uk/dp/B0C2ZYXNYQ?ref=ppx_yo2ov_dt_b_fed_asin_title)

**Daikin hardware**
S21

**Describe the bug**
I bought 2pcs of your [faikin board ](https://www.amazon.co.uk/dp/B0C2ZYXNYQ?ref=ppx_yo2ov_dt_b_fed_asin_title)and tried to power it with a usb-serial board. [Setup manual](https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Setup.md) does not shows [this version of the board](https://www.amazon.co.uk/dp/B0C2ZYXNYQ?ref=ppx_yo2ov_dt_b_fed_asin_title) which does not have gpio port as shown in the setup manual. 

_I need to power up the board and set up all necessary settings before connecting it to the Daikin. 
Could it be done? If so, how?_

**Expected behavior**
Working GPIO connection to the board.

revk, 2024-11-21:
Sorry do you have a picture of the board?

rici44, 2024-11-21:
> Sorry do you have a picture of the board?

Yes it is this board
[image]
[image]

rici44, 2024-11-21:
> If WiFI details are not correct it creates and access point to allow you to set new details.

OK. thanks

closed 2024-11-21 by revk

rici44, 2024-11-21:
> OK good, you can see the ⏚ and 12V as pins 5 and 4 on S21, power using those.

OK, thanks, powered up and running