revk/xESP32-Faikout #560: [BUG] Daikin Onecta not working

Taxxian, 2024-11-21:
**Faikin hardware**
Amazon 2 different versions

**Daikin hardware**
4 different Stylish Wall Device with S21 on a 5 MXM90 and a Daikin Home Hub

**Describe the bug**
I installed a Daikin Home Hub and Daikin ONECTA.
It was working fine for arround 3 days, then it suddenly stopped working.

Whenever I start the ONECTA App everything LOOKS fine, but all the values are frozen at the time it stopped working:
Temperatures
ON OFF state of all 4 Units
Energy Consuption
Calculated heat energy produced

Every single value is frozen even my Home Hub stays connected when I plug the power...
I restartet the App multiple times and there are no error messages. 

I have no Idea if that has anything to do with Faikin, but Daikin wont help me because I have Faikin and they usually do not answer anyways...

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

revk, 2024-11-22:
I have never heard of this before. But people do not usually use the Daikin WiFi modules and Onecta app once they have Faikin modules.

So does the onecta work if you do not have the Faikin connected.

Taxxian, 2024-11-22:
To find that out, I would have to dissassemble all the units and remove the Faikins... thats hours of work on a ladder I am not prepared to invest yet.

The problem is, that one can be legaly required to have Onecta running in germany, because the Daikin Home Hub limits the consumtion when less power is available. So it has to work.

Can you think of any setting that can lead to Problems with the Home Hub or Onecta?

Is there a way to reset Onecta. Daikin Software really seems to be bad.

revk, 2024-11-22:
Wow, I did not know that!

Does that mean you are legally required to have a working Internet connection as well. That is mental!

I'd suggest asking in the discussions and see what others have found. The way Faikin works is simply S21 messages, and it should not cause the app to behave any differently to any other S21 controller type device being connected.

If we have found a model which is not compatible, we should update the wiki.

But someone may have some other suggestions, I don't know.

Taxxian, 2024-11-23:
Daikin fixed the Problem.

I have no Idea if they just fixed my profile and the error will come back, or if they fixed the bug itself... I suspect the former.

Just like the JavaScript error they never wrote an answer, but fixed the problem exactly 24h after my mail... might be coincidence...

closed 2024-11-23 by revk