revk/xESP32-Faikout #561: [BUG] Faikin won't connect to Wifi

TheNetopyr, 2024-11-22:
**Faikin hardware**
Bought from Amazon: rev 2024-08-06
[image]
[image]

**Daikin hardware**
Connected to FTXM25A5V1B via S21 port

**Describe the bug**
De device works and I can connect in AP mode. It won't connect to my wifi (Mikrotik AP, 2,4ghz AX 1,5 meter away)

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in the power
2. Open the web interface on either a laptop or phone
3. Enter wifi credentials
4. Recieve Message "wifi did not connect, try again"

**Expected behavior**
A connected wifi

**Additional context**
Picture of the unit is exact same as currently installed without de extra snappable PCB board parts removed.
Wifi SSID name contains (capital)letters and a minus sign. Password contains (capital)letters, numbers and an ! Sign.
Additional question, can I update the firmware in AP mode?

revk, 2024-11-22:
My best guess: There is an issue in older software with long passwords. If you have a long password, change your wifi to a shorter one and get connected and update software, and then it will allow a longer password.

revk, 2024-11-22:
OK only other aspect is it has to be 2.4GHz

TheNetopyr, 2024-11-22:
Hi Revk, 
It is 2,4 ghz (I have seperated 5ghz on a different SSID).

revk, 2024-11-22:
OK not sure what to suggest, works on 2.4GHz for thousands of users. Your SSID and password sounds sane.

revk, 2024-11-23:
Well done. Good point on trying phone hotspot, I have done that before and should have suggested.