revk/xESP32-Faikout #565: [BUG]

seafurymike, 2024-11-28:
**Faikin hardware**
Purchased locally from UK stock

**Daikin hardware**
Connected via X35 and X50
Model FFQ35BV1B

**Describe the bug**
Unit will not come online. Checked wiring diagram and confirmed connectivity is correct from this link

https://github.com/revk/ESP32-Faikin/wiki/Wiring

**To Reproduce**
Plug it in

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
ERROR OUTPUT FROM MQTT STREAM BELOW

. Message 53 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"S21","cmd":"F1","noack":true,"value":"FE"}
QoS: 0 - Retain: false
Message 52 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A¬Tx¬Rx","badsum":"7A","data":"7505"}
QoS: 0 - Retain: false
Message 51 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A¬Rx","badsum":"7A","data":"7505"}
QoS: 0 - Retain: false
Message 50 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A¬Tx","badsum":"FE","data":"15EA15EA"}
QoS: 0 - Retain: false
Message 49 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A","badhead":true,"mismatch":true,"badrxlen":true,"data":"15EA"}
QoS: 0 - Retain: false
Message 48 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"S21¬Tx¬Rx","cmd":"F1","noack":true,"value":"01"}
QoS: 0 - Retain: false
Message 47 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"S21¬Rx","cmd":"F1","noack":true,"value":"01"}
QoS: 0 - Retain: false
Message 46 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"S21¬Tx","cmd":"F1","noack":true,"value":"FF"}
QoS: 0 - Retain: false
Message 45 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"S21","cmd":"F1","noack":true,"value":"F8"}
QoS: 0 - Retain: false
Message 44 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A¬Tx¬Rx","badsum":"F4","data":"75057505"}
QoS: 0 - Retain: false
Message 43 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A¬Rx","badsum":"7A","data":"7505"}
QoS: 0 - Retain: false
Message 42 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A¬Tx","badsum":"FE","data":"15EA15EA"}
QoS: 0 - Retain: false
Message 41 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"X50A","badhead":true,"mismatch":true,"badrxlen":true,"data":"15EA"}
QoS: 0 - Retain: false
Message 40 received on error/FaikinBroom/comms at 3:05 PM:
{"protocol":"S21¬Tx¬Rx","cmd":"F1","noack":true,"value":"01"}
QoS: 0 - Retain: false

revk, 2024-11-28:
If you know it is definitely S21 you can set nox50a, and the settings kit to invert tx and rx.

That will simplify the logs at least, as it tries many combinations.

It may then be worth power cycling the whole lot (air con too).

Then get some more logs from MQTT.

seafurymike, 2024-11-28:
wow. played with the protocol settings and it connected. it shows

Faikin-S3-MINI-N4-R2: ebf3c2b 2024-09-26T14:35:29 Altherma_S

so its no s21 or Nox50a??

Now, its connected but i can’t turn the init on via the web if i have MQTT connected?

revk, 2024-11-28:
I don't think Atherma allows control, just reporting of settings.

I'll have another look, but it is not S21 protocol.

seafurymike, 2024-11-28:
With debug on i don’t see anything come in.
With dump enabled I see the following only.

Message 344 received on info/FaikinBroom/rx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"S","dump":"53FF7F00040000000000000000000000002A"}
QoS: 0 - Retain: false
Message 343 received on info/FaikinBroom/tx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"S","dump":"0253AA"}
QoS: 0 - Retain: false
Message 342 received on info/FaikinBroom/rx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"T","dump":"54DC1DAC18271B0000001700000000000095"}
QoS: 0 - Retain: false
Message 341 received on info/FaikinBroom/tx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"T","dump":"0254A9"}
QoS: 0 - Retain: false
Message 340 received on info/FaikinBroom/rx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"U","dump":"550200000000410300000000000000000064"}
QoS: 0 - Retain: false
Message 339 received on info/FaikinBroom/tx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"U","dump":"0255A8"}
QoS: 0 - Retain: false
Message 338 received on info/FaikinBroom/rx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"S","dump":"53FF7F00040000000000000000000000002A"}
QoS: 0 - Retain: false
Message 337 received on info/FaikinBroom/tx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"S","dump":"0253AA"}
QoS: 0 - Retain: false
Message 336 received on info/FaikinBroom/rx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"T","dump":"54F41DAC18271B000000170000000000007D"}
QoS: 0 - Retain: false
Message 335 received on info/FaikinBroom/tx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"T","dump":"0254A9"}
QoS: 0 - Retain: false
Message 334 received on info/FaikinBroom/rx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"U","dump":"550200000000410300000000000000000064"}
QoS: 0 - Retain: false
Message 333 received on info/FaikinBroom/tx at 5:54 PM:
{"protocol":"Altherma_S","cmd":"U","dump":"0255A8"}
QoS: 0 - Retain: false

Is there anything specific I should enable to derive the info needed?

seafurymike, 2024-11-28:
This state info is presented at the start of listening

Message 354 received on homeassistant/climate/F0F5BD74290C/config at 5:58 PM:
{"unique_id":"F0F5BD74290C","dev":{"ids":["F0F5BD74290C"],"name":"FaikinBroom","sw":"ebf3c2b","mf":"RevK","cu":"http://FaikinBroom.local/"},"icon":"mdi:thermostat","avty_t":"state/FaikinBroom","avty_tpl":"{{value_json.up}}","pl_avail":true,"pl_not_avail":false,"~":"command/FaikinBroom","min_temp":16,"max_temp":32,"temp_unit":"C","temp_step":0.1,"temp_cmd_t":"~/temp","temp_stat_t":"state/FaikinBroom","temp_stat_tpl":"{{value_json.target}}"}
QoS: 0 - Retain: true
Message 353 received on state/FaikinBroom/status at 5:58 PM:
{"protocol":"Altherma_S","online":true,"autor":1.0,"autot":21.5,"auto0":"00:00","auto1":"00:00","autop":false}
QoS: 0 - Retain: true
Message 352 received on state/FaikinBroom at 5:58 PM:
{"ts":"2024-11-28T06:58:01Z","id":"F0F5BD74290C","up":true,"uptime":71,"mqtt-up":68,"app":"Faikin","version":"ebf3c2b","build-suffix":"-S3-MINI-N4-R2","build":"2024-09-26T14:35:29","flash":4194304,"rst":3,"mem":121928,"spi":2090472,"ssid":"<<REBOOTING NOW>>","bssid":"788A20F7972C","rssi":-48,"chan":11,"ipv4":"192.168.202.57","online":true,"target":21.50}
QoS: 0 - Retain: true

revk, 2024-11-28:
I am not sure how far Altherma development got, I thought we had basic state info. Select latest beta, if you have not already.

seafurymike, 2024-11-28:
Ok will try the beta.

seafurymike, 2024-11-28:
I upgraded to the beta version and tried to send an on command to the system (so hard to work out how to do this, could you please put examples in the manual for people to reference and make the link for the command easy to identify the options you can use)

So I sent the following to,power on the unit:

command/FaikinBroom/on

This is the error:

Message 86 received on Faikin/FaikinBroom at 8:48 PM:
{
    "protocol": "Altherma_S",
    "ts": "2024-11-28T09:48:00Z",
    "target": [
        20,
        22
    ],
    "online": true
}
QoS: 0 - Retain: false
Message 85 received on error/FaikinBroom/on at 8:48 PM:
{
    "ts": "2024-11-28T09:48:00Z",
    "description": "Setting cannot be controlled",
    "prefix": "command",
    "suffix": "on"
}
QoS: 0 - Retain: false
Message 84 received on error/FaikinBroom/control at 8:48 PM:
{
    "field": "power",
    "error": "Setting cannot be controlled"
}
QoS: 0 - Retain: false
Message 83 received on command/FaikinBroom/on at 8:48 PM:
QoS: 0 - Retain: false

then I tried to send a command with json format, I am not sure this is right how I formatted it, but got the following

Message 171 received on error/FaikinBroom/command at 9:00 PM:
{
    "ts": "2024-11-28T10:00:20Z",
    "description": "Extra value at top level",
    "prefix": "command",
    "target": "FaikinBroom",
    "suffix": "command",
    "payload": "\"power\":1"
}
QoS: 0 - Retain: false
Message 170 received on command/FaikinBroom/command at 9:00 PM:
"power":1
QoS: 0 - Retain: false

revk, 2024-11-28:
Yeh I don't think the interface for Atherma has a means to send any controls.

seafurymike, 2024-11-28:
I just want power on. Shame

revk, 2024-11-28:
I know there are some other people who have been working on this, but I think we concluded there is no "control" in the protocol even. If someone works it out I'll add to the code.

andi-christ, 2024-11-28:
@revk FYI I've had a batch manufactured and am distributing them in Australia. This is one of mine. @seafurymike send it back and I'll pass it on to someone else. Pls add your model to the list and indicate it's not compatible.